qmk/qmk_firmware #15178: Use Capslock toggle Chinese and English on macOS

hiimyusheng, 2021-11-16:
Hi,
I found that qmk firmware use capslock change input between chinese and english on macOS has  a issue
In a normal case, press capslock will toggle chinese and english, and english input will be lowercase
If want to input uppercase you just long press it, the input will be uppercase
but in qmk firmware, press capslock to change input from chinese to english, it also change english to uppercase,
it should be lowecase until i long press.
and in qmk firmware, long press will cause the input always be english, only if i long press again.
it seems like a bug, or just my settings is wrong?

filterpaper, 2021-11-18:
The feature you mentioned is based on timing detection of the operating system, not a bug. To enable better detection of Caps Lock on Mac OS, try adding the following delay to your `config.h`. Increment it slightly till the caps lock detection works for the feature you mentioned:
```c
#define TAP_CODE_DELAY 5
```
The value above worked did work correctly to tigger language change on tap, and upper case on long press.

hiimyusheng, 2021-11-18:
HI, 
my keyboard is gmmk pro
the default setting is `#define TAP_CODE_DELAY 10` 
I try to increce it, 11-13 the behavior as same as 10, above 13 it will not trigger language change 
when I tap the caps it always in English 
I also asked chinese keyboard community, there is people has the same issur same to me,
it seems like all qmk firmware on macOS's issue?

filterpaper, 2021-11-18:
Have you tried reducing to 5? I can confirm this not a QMK firmware issue with macOS issue because I have tested with QMK with MacOS and it works as described with the right timing.

drashna, 2021-11-18:
For capslock, it uses `TAP_HOLD_CAPS_DELAY`

lokher, 2022-02-08:
According to my test, it seems to be a bug of macOS handling NKRO input report. Turning off NKRO would solve it. Sending keys report via NKRO report interface only when more than 6 keys are pressed may be a better solution, current qmk firmware always sends key report via NKRO report interface when NKRO is enabled.

lokher, 2022-02-14:
I feedbacked this issue to Apple and am waiting for their response...

szcharlesji, 2022-04-08:
Seems that this issue has not been solved yet, for now, my solution is to remap caps lock to a macro in VIA.

Macro:`{KC_LCTL,KC_SPC}`

This switches your input source only. But the tradeoff is that you cannot use it for a caps lock anymore.

antonu17, 2022-05-02:
> According to my test, it seems to be a bug of macOS handling NKRO input report. Turning off NKRO would solve it. Sending keys report via NKRO report interface only when more than 6 keys are pressed may be a better solution, current qmk firmware always sends keys report via NKRO report interface when NKRO is enabled.

Indeed. I cloned https://github.com/Keychron/qmk_firmware and flashed `qmk flash -kb keychron/q1/rev_0107 -km default` with `NKRO_ENABLE = no` and Caps Lock is now working properly for switching layouts.

Agilato, 2022-07-23:
> > According to my test, it seems to be a bug of macOS handling NKRO input report. Turning off NKRO would solve it. Sending keys report via NKRO report interface only when more than 6 keys are pressed may be a better solution, current qmk firmware always sends keys report via NKRO report interface when NKRO is enabled.
> 
> Indeed. I cloned https://github.com/Keychron/qmk_firmware and flashed `qmk flash -kb keychron/q1/rev_0107 -km default` with [`NKRO_ENABLE = no`](https://github.com/Keychron/qmk_firmware/blob/dc2d0939b916b013cb73f1f761f560eb42c53bc8/keyboards/keychron/q1/rev_0107/rules.mk#L15) and Caps Lock is now working properly for switching layouts.
> 
> Before doing this I also tried to set `TAP_CODE_DELAY` to `5` and `TAP_HOLD_CAPS_DELAY` to `200` as recommended in https://github.com/qmk/qmk_firmware/blob/master/docs/config_options.md - these parameters didn't help.
> 
> cc @lalalademaxiya1 cc @fauxpark

Used this method for my US ANSI Knob Q1 V2 and after flashing everything seems to be working fine and NKRO is disabled but VIA no longer detects the keyboard, noticed it when my RGB was no longer controlled by FN+Knob and decided to check keymap. Is there any deeper manual about settings in rules.mk file?

fauxpark, 2022-07-23:
VIA won't work when you flash the `default` keymap.

Agilato, 2022-07-23:
> VIA won't work when you flash the `default` keymap.

Make sense but with VIA keymap it throws a error about EPROM size

`Compiling: quantum/dynamic_keymap.c                                                                quantum/dynamic_keymap.c:47:13: note: '#pragma message: 2047 > ((1024) - 1)'
   47 | #    pragma message STR(DYNAMIC_KEYMAP_EEPROM_MAX_ADDR) " > " STR((TOTAL_EEPROM_BYTE_COUNT - 1))
      |             ^~~~~~~
quantum/dynamic_keymap.c:48:6: error: #error DYNAMIC_KEYMAP_EEPROM_MAX_ADDR is configured to use more space than what is available for the selected EEPROM driver
   48 | #    error DYNAMIC_KEYMAP_EEPROM_MAX_ADDR is configured to use more space than what is available for the selected EEPROM driver
      |      ^~~~~
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:456: .build/obj_keychron_q1_rev_0107_via/quantum/dynamic_keymap.o] Error 1
gmake: *** [Makefile:413: keychron/q1/rev_0107:via:flash] Error 1`

lokher, 2022-07-23:
> Make sense but with VIA keymap it throws a error about EPROM size
>
> `Compiling: quantum/dynamic_keymap.c quantum/dynamic_keymap.c:47:13: note: '#pragma message: 2047 > ((1024) - 1)' 47 | # pragma [...]

Did use `playground` branch of Keychron fork?

nitrojector, 2022-07-23:
> > > VIA won't work when you flash the `default` keymap.
> > 
> > 
> > Make sense but with VIA keymap it throws a error about EPROM size
> > `Compiling: quantum/dynamic_keymap.c quantum/dynamic_keymap.c:47:13: note: '#pragma message: 2047 > ((1024) - 1)' 47 | # pragma message STR(DYNAMIC_KEYMAP_EEPROM_MAX_ADDR) " > " STR((TOTAL_EEPROM_BYTE_COUNT - 1)) | ^~~~~~~ quantum/dynamic_keymap.c:48:6: error: #error DYNAMIC_KEYMAP_EEPROM_MAX_ADDR is configured to use more space than what is available for the selected EEPROM driver 48 | # error DYNAMIC_KEYMAP_EEPROM_MAX_ADDR is configured to use more space than what is available for the selected EEPROM driver | ^~~~~ [ERRORS] | | | gmake[1]: *** [builddefs/common_rules.mk:456: .build/obj_keychron_q1_rev_0107_via/quantum/dynamic_keymap.o] Error 1 gmake: *** [Makefile:413: keychron/q1/rev_0107:via:flash] Error 1`
> 
> Did you compile with `playground` branch of Keychron fork?

Will compiling using any other keymaps listed [here](https://github.com/Keychron/qmk_firmware/tree/dc2d0939b916b013cb73f1f761f560eb42c53bc8/keyboards/keychron/q1/rev_0100/keymaps) (excluding via) work with VIA?

Agilato, 2022-07-23:
> Did you compile with `playground` branch of Keychron fork?

Thank you, playground branch has compiled fine.

closed 2022-11-22 by github-actions[bot]