qmk/qmk_firmware #15179: [Feature Request] Display/Store/Use name of effect instead of its index

blueyed, 2021-11-16:
When changing an effect you will see e.g. "rgb matrix mode [EEPROM]: 8" in the debug output (via `hid_listen` in my case).
This however does not allow to easily map it to the effective effect, since it depends on the ones you have enabled/disabled etc.
Therefore it would be better if it would display "rgb matrix mode [EEPROM]: 8 (TYPING_HEATMAP)" or similar.

The actual name should also get displayed with `eeconfig_debug_rgb_matrix`, which is only used inside of `rgb_matrix.c`, but should eventually get exported to be usable from keymaps etc.

I could imagine adding a new variable for this, maybe into `rgb_matrix_config`, and setting it in the `RGB_MATRIX_EFFECT` macro then [1], or better add a new case into where it gets changed (`rgb_matrix_mode_eeprom_helper`).

This could also get extended to replace the existing `mode` integer, so that restoring/using a saved mode would work with a new list of modes (i.e. mapping it by name instead of id/int when restoring/reading the config).

This also applies to the LED matrix feature as far as I can see, so should get factored out accordingly probably.

1: https://github.com/qmk/qmk_firmware/blob/c48e9f9e77a8f8c722957ef306280014b098758d/quantum/rgb_matrix/rgb_matrix.c#L362-L365

## Feature Request Type

- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)

drashna, 2021-11-16:
I believe that tzarc had some ideas on how to better handle this, especially in regards to VIA/XAP.

tzarc, 2021-11-16:
The main reason this doesn't exist is because the names take up flash space, and AVR builds basically have none available.

If you really want it, you can add this to your keymap:
https://github.com/tzarc/qmk_build/blob/f8265fc900b990a23a10848b5def70bd3a3b2860/tzarc-djinn/keymaps/tzarc/keymap.c#L167-L208

You can then call `rgb_matrix_name(rgb_matrix_config.mode)` to get the current effect's name.

closed 2022-06-18 by github-actions[bot]