qmk/qmk_firmware #15216: Laptop-style Numlock

donmor, 2021-11-19:
I'm using a 75% keyboard that have a laptop-like layout, which has no numpad. These days I'm trying to implement a numpad using VIA but with no luck.

The keymap I'm trying to make is supposed to be:
```
7 8 9 0                     7 8 9 *
 U I O P         <--->       4 5 6 /
  J K L ;     Fn or NumLk     1 2 3 -
   M   . ?                     0   . +
```
Anyone have ideas?

donmor, 2021-11-19:
BTW It should get the state of NumLk of current system when plugged in and change the layout, and should also be affected by switching NumLk on another keyboard.

peepeetee, 2021-11-19:
Looks fairly simple, should just require 1 extra layer. What are the problems you are encountering?

donmor, 2021-11-19:
Just using layers messes up the NumLk state. For example, plugging the keyboard in when the PC's NumLk is on, and the keys still in normal mode, and need the fake NumLk to be pressed.

donmor, 2021-11-19:
So are macros able to detect the state of the NumLk and return different key codes? Like an if structure.

peepeetee, 2021-11-19:
Yes, check if  `host_keyboard_led_state().num_lock == true`

If yes, then the numlock is on and do the numlocked stuff
If no continue as normal

donmor, 2021-11-19:
So I need to put the C code in a macro? I'm using VIA

peepeetee, 2021-11-19:
If you want to get the state of NumLk of current system, then you would need the C code I posted above and put it in your own keymap. To build a keymap, see https://docs.qmk.fm/#/newbs_building_firmware

If you want to avoid this and just use VIA for this, I don't think there's a way to do it. Best you can do would be having a second layer and switching to it with a key

peepeetee, 2021-11-19:
您非得要和电脑同步吗？ 不和它同步会简单很多
以及我们有个群，要加吗？

closed 2022-06-18 by github-actions[bot]