qmk/qmk_firmware #15217: [Bug] When I2C and split keyboard is used, it is not recognized as USB HID device.

7-rate, 2021-11-19:
## Describe the Bug

The split keyboard that uses I2C as the protocol for left-right communication does not get recognized as USB HID.
The problem does not occur in 0.13.38(9472c4b) , but it occurs in 0.14.0(f061ca49).
I have reproduced this bug with at least pistachio and splitty keyboards.

## System Information

 - Keyboard:  pistachio
 - Operating system: windows10
 - AVR GCC version: 10.0.1
 - ARM GCC version: 10.0.1
 - QMK Firmware version: 0.14.0
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

The following are the results I checked with USB Device Tree View V3.5.3 when the problem was occurring.
```text

    =========================== USB Port12 ===========================

Connection Status        : 0x02 (Device failed enumeration)
Port Chain               : 4-12
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x0C (Port 12)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#5&1171eda9&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x04 (Port 4)
 -> CompanionPortChain   : 4-4

Device Manager Problem   : 43 (CM_PROB_FAILED_POST_START)
Used Endpoints           : 0

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : 不明な USB デバイス (デバイス記述子要求の失敗)
Device ID                : USB\VID_0000&PID_0002\6&104FED9E&0&12
Hardware IDs             : USB\DEVICE_DESCRIPTOR_FAILURE
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0037 (GUID_DEVCLASS_USB)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Enumerator               : USB
PDO                      : \Device\USBPDO-12
Location Info            : Port_#0012.Hub_#0001
Manufacturer Info        : (標準 USB ホスト コントローラー)
Capabilities             : 0x64 (Removable, SilentInstall, RawDeviceOK)
Status                   : 0x01806400 (DN_HAS_PROBLEM, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 43 (CM_PROB_FAILED_POST_START)
Address                  : 12
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D3 (supported: D0, D3, wake from D0)

        ---------------- Connection Information ---------------
Connection Index         : 0x0C (Port 12)
Connection Status        : 0x02 (DeviceFailedEnumeration)
Current Config Value     : 0x00 (Configuration 0)
Device Address           : 0x00 (0)
Is Hub                   : 0x00 (no)
Device Bus Speed         : 0x01 (Full-Speed)
Number Of Open Pipes     : 0x00 (0 pipes to data endpoints)
Data (HexDump)           : 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
                           00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 02   ................
                           00 00 00                                          ...

        --------------- Connection Information V2 -------------
Connection Index         : 0x0C (12)
Length                   : 0x10 (16 bytes)
SupportedUsbProtocols    : 0x03
 Usb110                  : 1 (yes, port supports USB 1.1)
 Usb200                  : 1 (yes, port supports USB 2.0)
 Usb300                  : 0 (no, port not supports USB 3.0) -> but Companion Port 4-4 does
 ReservedMBZ             : 0x00
Flags                    : 0x00
 DevIsOpAtSsOrHigher     : 0 (Device is not operating at SuperSpeed or higher)
 DevIsSsCapOrHigher      : 0 (Device is not SuperSpeed capable or higher)
 DevIsOpAtSsPlusOrHigher : 0 (Device is not operating at SuperSpeedPlus or higher)
 DevIsSsPlusCapOrHigher  : 0 (Device is not SuperSpeedPlus capable or higher)
 ReservedMBZ             : 0x00
Data (HexDump)           : 0C 00 00 00 10 00 00 00 03 00 00 00 00 00 00 00   ................

    ---------------------- Device Descriptor ----------------------
bLength                  : 0x00 (0 bytes)

    ----------------- Device Qualifier Descriptor -----------------
Error                    : ERROR_GEN_FAILURE

      -------------------- String Descriptors -------------------
String descriptors are not available  (because the device has problem code CM_PROB_FAILED_POST_START)

```

7-rate, 2021-11-26:
The phenomenon occurs when the left and right sides are not connected.

drashna, 2021-11-26:
This is a known issue with the newer split code, and it is being looked into, IIRC.

For now, the solution is to have both sides plugged in.

tim-clifford, 2021-12-30:
if it helps, i did some `git bisect`ing and I believe the offending commit is ef92c9e

doctorkelp, 2022-01-20:
Any chance it's related to https://github.com/qmk/qmk_firmware/pull/13894?

zvecr, 2022-01-20:
> Any chance it's related to #13894?

Almost zero chance, given the code ended up functionally identical.

Kriechi, 2022-03-05:
I just stumbled upon the same problem - this is still an issue on 0.16.3.

`#define USE_SERIAL` works and connects properly, also `qmk console` works.
`#define USE_I2C` doesn't work at all - not recognized as HID and no `qmk console` either.

danielvandenberg95, 2022-04-26:
Just to help others find this issue more easily (took me forever to find it).
[image]

This gives the message: Unknown USB (Device Descriptor Request Failed)

daskygit, 2022-05-04:
This I2C split USB enumeration issue has been fixed by #16647. It's already been merged into develop and should land in master this breaking change cycle. If you want to give it a test checkout the develop branch.

7-rate, 2022-05-14:
I have confirmed that the problem has been resolved on the develop branch.

closed 2022-07-04 by drashna