qmk/qmk_firmware #15219: [Feature Request] Add CAPS(KC1,KC2) / SCRL(KC1,KC2) / NMLK(KC1,KC2)

donmor, 2021-11-19:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Each of these 3 functions sends KC1 if a certain lock is activated, otherwise send KC2. Useful to add keypad to non-full keyboards.

fauxpark, 2021-11-19:
Quantum keycodes which contain two basic keycodes are not possible with the way the keycode system works currently. See this page: https://docs.qmk.fm/#/mod_tap?id=caveats

donmor, 2021-11-19:
I see... So is there a alternative way to do this, like some kinds of macros?

donmor, 2021-11-19:
BTW it'll be even better if it can be deployed by VIA :)

filterpaper, 2021-11-21:
You can use the macro examples to send:
```c
host_keyboard_led_state().caps_lock ? tap_code16(<kc1>) : tap_code16(<kc2>);
```

donmor, 2021-11-21:
Thanks, and can I use this in VIA?

filterpaper, 2021-11-21:
Nope, you will need to build your own firmware with custom code.

donmor, 2021-11-21:
You mean defining this in a macro in C code so I can use it in VIA?

filterpaper, 2021-11-21:
You can define a "custom keycode" that you can probably use in VIA.

closed 2022-06-18 by github-actions[bot]