qmk/qmk_firmware #15225: Security and compatibility concerns about the udev rules

Andrew-Fahmy, 2021-11-20:
I installed the udev rules to `/etc/udev/rules.d/` as recommended by `qmk doctor`. Afterwards this rule was causing problems:
```KERNEL=="hidraw*", MODE="0660", GROUP="plugdev", TAG+="uaccess", TAG+="udev-acl"```

In my case I it was messing with the way that the [xpadneo](https://github.com/atar-axis/xpadneo) driver uses the hidraw devices. When I mentioned this issue to that projects maintainer he also brought up how it could be a security issue:

> It allows any local process in your session to spy on any hidraw device (even when unrelated to your session), which is almost every input device in the world, thus allowing to record keypresses, pin entries, or more generally: secrets.
[(Full context)](https://github.com/atar-axis/xpadneo/issues/303#issuecomment-973854335)

For now I can just comment out that rule and everything works fine but going forward maybe there is a better approach.

fauxpark, 2021-11-20:
https://github.com/qmk/qmk_firmware/pull/12980

Andrew-Fahmy, 2021-11-20:
Oh thanks for linking that, I didn't know there was another repo just for the udev rules.

I have installed the rules from that project but the doctor command is telling me they are out of date. How does it verify if your rules are correct?

closed 2022-06-18 by github-actions[bot]