qmk/qmk_firmware #15251: [Bug] upstream merge on a fork

jackokring, 2021-11-21:
## Describe the Bug
A qmk compile no longer works.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

V 1.7.2 worked fine. I suppose I'll have to rollback until it works.

jackokring, 2021-11-21:
Ok. So on the fork make a branch based on the rollback and git checkout that new branch. Hopefully the upstream master gets better.

Maybe try an occasional merge of the branch into master to check on the build system process.

I might perform a full chop back of other designs in the branch. Decluttering.

Perhaps a fork proxy of the branch for a max function dz60 competition?

`jackokring/qmk_firmware` branch `Gooo-Mk-II` and start an interest "issue" ...?

tzarc, 2021-11-21:
```
❱ qmk compile -kb planck/rev6 -km default                                                                                                                                                                                                                                                                                 Ψ Compiling keymap with gmake --jobs=40 --output-sync=target planck/rev6:default                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              QMK Firmware 0.14.32                                                                                                                                                                                                                                                                                                      Making planck/rev6 with keymap default                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
<<<snip>>>
Linking: .build/planck_rev6_default.elf                                                             [OK]
Creating load file for flashing: .build/planck_rev6_default.hex                                     [OK]

Size after:
   text    data     bss     dec     hex filename
      0   53522       0   53522    d112 .build/planck_rev6_default.hex

Creating binary load file for flashing: .build/planck_rev6_default.bin                              [OK]
Copying planck_rev6_default.bin to qmk_firmware folder                                              [OK]
(Firmware size check does not yet support STM32F303; skipping)```

Running `qmk compile` on `master` works fine. I'm not sure what your issue is?

jackokring, 2021-11-22:
I merged master into a downstream repository `jackokring/qmk_firmware` and building the keyboard code stopped working. I attempted install of the tools again `qmk setup` and after a few missing libraries, it said all was OK. The strange thing was that `qmk compile` complained about only the words `config`, `setup` and `install` were available. `compile` was not a recognized thing to follow `qmk`.

I'm assuming there is a merge issue which as I have not touched the core code is strange.

jackokring, 2021-11-27:
Found out `~/.local` had some python cache stuff in. Deleted it. Now works.

closed 2021-11-27 by jackokring