qmk/qmk_firmware #15257: How to achieve the power-on is the all-key non-impact state?

yanluoxingchen, 2021-11-22:
Main control model: 32U4
PCB model :GH60 SATAN
BOOTMAGIC_ENABLE = lite yes I have tried this, but it is not written to EEPROM
1. What requirements must be met at least to achieve power-on all-key non-flushing? Are there any restrictions on the BOOTLOADER?
2. What is the firmware built with BOOTMAGIC_ENABLE mentioned in NKRO

peepeetee, 2021-11-22:
在rules.mk里试一试 NKRO = yes

yanluoxingchen, 2021-11-22:
> 在 rules.mk 里试一试 NKRO = 是

我试过这个这个功能是可以用的  但是无法记录到EEPROM里,6键无冲和全键无冲也能切换

peepeetee, 2021-11-22:
[image]

peepeetee, 2021-11-22:
您試試在 config.h裡加上 
`#define FORCE_NKRO`

yanluoxingchen, 2021-11-22:
> 您試試在 config.h裡加上 `#define FORCE_NKRO`

非常感谢，非常感谢大佬谢谢谢谢  可以了

closed 2021-11-22 by yanluoxingchen

peepeetee, 2021-11-22:
考虑一下来我们的discord群组？ https://discord.com/invite/Uq7gcHh

yanluoxingchen, 2021-11-22:
> 考虑一下来我们的discord群组[https://discord.com/invite/Uq7gcHh？](https://discord.com/invite/Uq7gcHh)

去大佬邀请是荣幸啊