qmk/qmk_firmware #15267: [Feature Request] Support GMMK PRO's "Cycle through Active RGB Zone"

dxps, 2021-11-22:
### Description

Dear QMK community,

Any change to add the _Cycle through Active RGB Zone_ key option of GMMK Pro to QMK Configurator, please?

As you can find in the [Product Guide](https://www.pcgamingrace.com/pages/product-guides?guide=GMMK_PRO#Quick-Start-Guide), section _RGB LED Light Settings_, `FN + 1` allows choosing whether both (front+side) or one of them to get lights on. [This reddit thread](https://www.reddit.com/r/glorious/comments/nhtvvw/gmmk_pro_how_to_turn_off_all_lights_except_side/) confirms its existence on their firmware.

I'm particularly interested in just having the side lights on.

Notes:
1. If it's already there, just enlight me. I search though existing issues before raising this one.
2. If this request should actually first be raised to QMK repo, I'll do it.

Thanks!

dxps, 2021-12-04:
@yanfali Thanks for moving it into the proper repo. If there anything I can do to help, please tell me.

drashna, 2021-12-13:
I believe that this can be done by using the rgb matrix set flags function.

dxps, 2021-12-13:
@drashna 
You mean using `rgb_matrix_set_flags` function.
Thanks for the hint! :pray: 

Meanwhile, without further details, I assume that I should try to understand how to use it based on concrete examples
such as [/keyboards/moonlander/moonlander.c#L436](https://github.com/qmk/qmk_firmware/blob/f718a10889e6adf33f3fc2f41b61cad7fe9e0c2e/keyboards/moonlander/moonlander.c#L436).

closed 2022-07-13 by github-actions[bot]