qmk/qmk_firmware #15273: [Bug] RGB matrix lighting broken on K-type due to incorrect LED_FLAGS

Andrew-Fahmy, 2021-11-23:
On my K-type the rgb matrix lighting stops working if I unplug it or reflash it. I have to manually set the led flags like this:
```c
void keyboard_post_init_user(void) {
    rgb_matrix_set_flags(0xFF);
}

```
It looks like the `rgb_matrix_config.flags` defaults to 0 and then overwrite the `rgb_effect_params.flags` [here](https://github.com/qmk/qmk_firmware/blob/97a32eedf1cebc1fdd771d0c67ee8320c06e8d99/quantum/rgb_matrix/rgb_matrix.c#L340)

```c
    if (rgb_effect_params.flags != rgb_matrix_config.flags) {
        rgb_effect_params.flags = rgb_matrix_config.flags;
        rgb_matrix_set_color_all(0, 0, 0);
    }
```
I updated my keyboard for the first time in a while and when I did this was the issue that I was getting. Could this be something specific to my keyboard or my config?

filterpaper, 2021-11-30:
You can try clearing the EEPROM, `rgb_matrix_config.flags` should use `LED_FLAG_ALL` as default:
https://github.com/qmk/qmk_firmware/blob/master/quantum/rgb_matrix/rgb_matrix.c#L152

Andrew-Fahmy, 2021-12-02:
It seems like that function is not being called. If I call it manually from `keyboard_post_init_user` it works. When looking into the the `rgb_matrix_config` there is a comment before the flags that says `EECONFIG needs to be increased to support this` [here](https://github.com/qmk/qmk_firmware/blob/b6d9a8026be574d13eff2cb6a06978c02a56c567/quantum/rgb_matrix/rgb_matrix_types.h#L91). My guess is that the first three variables are stored in EEPROM so they are correctly loaded, but the flags and speed are not so they default to 0. 

If that is the case it could be solved by defaulting the flags to `LED_FLAG_ALL` statically:
```c
rgb_config_t rgb_matrix_config = { .flags = LED_FLAG_ALL };
```
This is a bit of a hack but fixed my keyboard for now.

Andrew-Fahmy, 2021-12-05:
It looks like this is a known issue. https://github.com/qmk/qmk_firmware/issues/13164 talks about it more. For now I will implement a similar solution to the one on the whitefox.

closed 2021-12-05 by Andrew-Fahmy