qmk/qmk_firmware #15331: [Bug] 0.14.33 breaks backlight levels

karlshea, 2021-11-28:
## Describe the Bug

I have a higher number of backlight levels (20) so it's more obvious, but even a smaller number shows this behavior: increasing/decreasing the backlight levels jumps the light intensity all over the place (e.g. starting from backlight level 0 -> intensity 0, 1 -> intensity 50 (estimated), 2 -> intensity 10, 3 -> intensity 200, 4 -> intensity 50, 5 -> intensity 190).

0.14.32 works normally. If I checkout the first commit (b47bdee93c5dc39353477126abd344476c736132 ) after that tag it starts to break.

## System Information

 - Keyboard:  KBDfans KBD75
   - Revision (if applicable):  rev2
 - Operating system: macOS
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Building using latest qmkfm/qmk_cli Docker image.

tzarc, 2021-11-28:
Do you know how to do a `git bisect`?

karlshea, 2021-11-28:
I didn't, but now I do, it says e0a5056963beddc3430dd3c8372858288dc85809 is the first bad commit.

karlshea, 2021-11-28:
It's something to do with `cie_lightness()`. If I revert just those changes it works fine too.

karlshea, 2021-11-28:
Figured it out, it's losing bits doing division.

PR #15344 (2021-11-29): Fix bit loss in cie_lightness() when doing division to resolve #15331
## Description

Bit loss during division breaks static backlight levels.

@Duckle29 since you authored the PR that made these changes can you confirm this doesn't break what you were trying to accomplish?

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #15331

## Checklist

- [x] My code follows the code style of this project:  [... cut]
commit 84b8cdc1 (2021-11-29): Fix bit loss in cie_lightness() when doing division to resolve #15331 (#15344)

* Fix bit loss in cie_lightness() when doing division.

* Use the right types

* Format
closed 2021-11-29 by zvecr