qmk/qmk_firmware #15345: [Bug] RGBLIGHT_EFFECT_TWINKLE is nonfunctional on ARM

bcat, 2021-11-28:
## Describe the Bug

Repo'd on latest QMK master today. It seems the `RGBLIGHT_EFFECT_TWINKLE` animation mode doesn't work at all on ARM, or at least not on the CannonKeys Instant60. The LEDs all stay off when that mode is active.

I saw some references to this on the QMK Discord, but didn't see an issue filed for it yet. (I also didn't test an AVR keyboard, but it sounds like the twinkle mode works correctly there.)

## System Information

 - Keyboard: `cannonkeys/instant60`
   - Revision (if applicable): N/A
 - Operating system: Windows 10 Version 21H2 (OS Build 19044.1387)
 - AVR GCC version: `avr-gcc (AVR-GCC) 8.5.0`

 - ARM GCC version: `arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]`

 - QMK Firmware version: `0.15.0`

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

osleg, 2021-12-02:
Even more, it looks like twinkle is not functional. period.

drashna, 2021-12-03:
twinkle work on AVR just fine.  But not working on ARM is a known issue. 

I've been using this horrible hack to get it working on both:
https://github.com/drashna/qmk_firmware/commit/2ac3e142ff77516d3c19d8322fee066480529f51

drashna, 2022-12-09:
oh, @spidey3

yiancar, 2022-12-09:
This at ARM has to do with the fact that we are comparing rand() which is a 32bit int to a clipped 8bit number
`rand() < scale((uint16_t)RAND_MAX * RGBLIGHT_EFFECT_TWINKLE_PROBABILITY, 127 + rgblight_config.val / 2)`

This statement is extremly rare to be true. Can the rational behind changin this from:
`rand() < RAND_MAX * RGBLIGHT_EFFECT_TWINKLE_PROBABILITY` be explained? Is this to scale how much the effect twinkles in less bright values?

the main reason this breaks is that RAND_MAX in avr is 0x7FFF wheres in arm it is 7FFFFFFF

drashna, 2022-12-09:
would it be reasonable to do something like: 
```c
        } else if ((rand() % 0x7FFF) < scale((uint16_t)RAND_MAX * RGBLIGHT_EFFECT_TWINKLE_PROBABILITY, 127 + rgblight_config.val / 2)) {
```

spidey3, 2022-12-09:
Thanks for pointing this out.
I should have some time to look into it this weekend.

yiancar, 2022-12-12:
@drashna we could `(rand() % (RAND_MAX >> 8))` because otherwise we will always be on the lowside of probability for arm.
If @spidey3 can explain the intention of scaling with brightness then we can explore the above or if the brightness scaling doesnt make a difference we can revert to what was before (previews effect coding just for this line) for the probability math.

spidey3, 2022-12-15:
I finally had some time today to look into this. The intent here was to decrease the probability of triggering an LED to start brightening, and also reduce the maximum length of a "twinkle", when brightness is lower. How did I come up with that? It was just experimentation; it looked better (especially with the room lights off).

We could go with @drashna's suggestion above, as it does work, but I am somewhat perturbed by the hard-coded constant...

spidey3, 2022-12-15:
Actually, looking at this some more, it's actually assuming that RAND_MAX <= 0xFF, which isn't the case even on AVR!

closed 2022-12-15 by spidey3