qmk/qmk_firmware #15347: A bit of a problem with the full key

yanluoxingchen, 2021-11-29:
A bit of a problem with the full key
Main control model: AT90USB1286
The problem is that I added #define FORCE_NKRO to the config.h file
Now the full key can be used normally without flushing, but there is no way to switch 6NKRO and other debugging keys cannot be used
I would like to ask some questions:
1. Check whether the BOOTLOADER affects NKRO functions
2. If BOOTMAGIC_ENABLE = lite NKRO_ENABLE = yes, no #define FORCE_NKRO NKRO 6 button is added to config.h. And other debugging functions are not available

drashna, 2021-11-29:
`MAGIC_HOST_NKRO`, `MAGIC_UNHOST_NKRO` and `MAGIC_TOGGLE_NKRO` still work, but NRKO will be forced on each time the board starts.   It's one of the reasons I don't recommend using `FORCE_NKRO`

yanluoxingchen, 2021-11-29:
> `MAGIC_HOST_NKRO`，并且仍然有效，但每次启动板时，NRKO 都将强制打开。这是我不建议使用的原因之一`MAGIC_UNHOST_NKRO``MAGIC_TOGGLE_NKRO``FORCE_NKRO`
If #define FORCE_NKRO is not used, how can NKRO be automatically turned on
What I want to do is power on the default board in the NKRO state and I can switch it to the 6NKRO state with key combination

yanluoxingchen, 2021-11-29:
> `MAGIC_HOST_NKRO`, `MAGIC_UNHOST_NKRO` and `MAGIC_TOGGLE_NKRO` still work, but NRKO will be forced on each time the board starts. It's one of the reasons I don't recommend using `FORCE_NKRO`

I also don't really want to use #define FORCE_NKRO
However, the normal declaration of NKRO_ENABLE = yes in rules.mk cannot be recorded in EEPROM.
Using #define FORCE_NKRO is also a must

drashna, 2021-11-29:
The magic keycodes write to eeprom, so the settings are persistent. 

Force NKRO just sets it to be enabled on startup.  you can turn it off, afterwards.

yanluoxingchen, 2021-11-29:
> 
I understand that writing to an EEPROM is permanent, but I can't write to an EEPROM now so I used this method instead

closed 2021-12-01 by yanluoxingchen