qmk/qmk_firmware #15375: Compiling for 1up60hse does not work

capaldo, 2021-12-01:
Running:

`qmk info -kb 1upkeyboards/1up60hse -km default`

Results in:

```
usage: qmk info [-h] [-r] [--ascii] [-f FORMAT] [-m] [-l] [-km KEYMAP] [-kb KEYBOARD]
qmk info: error: argument -kb/--keyboard: invalid keyboard_folder value: '1upkeyboards/1up60hse'
```

tzarc, 2021-12-01:
Can't reproduce the `qmk info` problem:
```
% qmk info -kb 1upkeyboards/1up60hse -km default
Keyboard Name: 1up60hse
Manufacturer: 1upkeyboards
Website:
Maintainer: QMK Community
Keyboard Folder: 1upkeyboards/1up60hse
Layouts: LAYOUT_60_ansi
Processor: atmega32u4
Bootloader: atmel-dfu
```

As for attempting to build all, you probably want `qmk multibuild`:
```
% qmk multibuild -j40
Build 0xcb/static:default                                              [OK]
Build 0xc7/61key:default                                               [OK]
Build 2key2crawl:default                                               [OK]
Build 30wer:default                                                    [OK]
Build 40percentclub/polyandry/teensy2:default                          [OK]
Build 40percentclub/polyandry/promicro:default                         [OK]
Build 40percentclub/i75/promicro:default                               [OK]
Build 40percentclub/4pack:default                                      [OK]
Build 0_sixty:default                                                  [OK]
Build 40percentclub/5x5:default                                        [OK]
Build 1upkeyboards/sweet16/v2/promicro:default                         [OK]
Build 1upkeyboards/sweet16/v1:default                                  [OK]
Build 3w6/rev1:default                                                 [OK]
Build 3w6/rev2:default                                                 [OK]
Build 40percentclub/sixpack:default                                    [OK]
Build 1upkeyboards/super16:default                                     [OK]
Build 40percentclub/mf68:default                                       [OK]
Build 40percentclub/ut47:default                                       [OK]
Build 45_ats:default                                                   [OK]
Build 3keyecosystem/2key2:default                                      [OK]
<snip>
```

capaldo, 2021-12-09:
So after running multibuild, it started working for specific keyboards. Most likely this is the result of a user error on my part.

closed 2021-12-09 by capaldo