qmk/qmk_firmware #15397: Can I achieve LCTL(KC_TRNS)?

brhigginsuk, 2021-12-04:
Fairly new to QMK.

I have a key that I want to act as left control when combined with most other keys on the keyboard, but behave differently for a few specific combinations. In my mind, this should be achieved by having a second layer with all keys set to `LCTL(KC_TRNS)`, except for the few that should behave differently.
However, LCTL(KC_TRNS) appears to do nothing when flashed to my keyboard.

I could copy-paste the whole of the first layer and add `LCTL(..)` to everything, but then any change to my base layer needs to be manually propogated.

Is this achievable?
Thanks.

brhigginsuk, 2021-12-04:
Never mind, on a subsequent attempt this works fine. Must have had something else wronog somewhere!

drashna, 2021-12-24:
You mean so that it uses the keycode from the layer below it?

If so, then, no, that is not supported.  But you may be able to use `LM` for the layer key instead of this.

brhigginsuk, 2021-12-26:
Yeah, that's what I meant. I wanted to use Caps as Ctrl, except for when combined with HJKL (for system-wide vim-like movement). Ctrl plus any of those would require hitting the actual Ctrl key. Not a big deal though, thanks for the response.

closed 2021-12-26 by brhigginsuk