qmk/qmk_firmware #15400: [Bug] ymdk_np21 works unreliably after update

gudvinr, 2021-12-05:
I used some older version of QMK for probably a year or so and decided to upgrade recently.  
Everything worked without issues before update but then suddenly I have multiple problems with QMK.

I didn't pay attention to which version was working because I didn't think I could get malfunctioning keyboard after update.

## Describe the Bug

1. When I press any configured key (e.g. `C(KC_F13)`, `C(KC_V)` and both tapdance actions for `ACTION_TAP_DANCE_DOUBLE(S(KC_F16),  C(S(KC_F16)))` are ones that I use often) it can sometimes be ignore.  
  I don't see any correlation with time between keypresses in any case.
2. Sometimes I get multiple events fired. For example. one of the keys is `A(KC_F4)` and it could just happily wipe out all the windows I have.

## System Information

 - Keyboard:  ymdk_np21
 - Operating system: Arch Linux
 - AVR GCC version: one inside `qmkfm/qmk_cli` container

 - ARM GCC version: one inside `qmkfm/qmk_cli` container

 - QMK Firmware version:  
    - 0.15.3-b98280bf5 right now
    - noticed at 0.14.32 (-ish)
    - got keyboard at 2020-06-02 so not older than that

 - Any keyboard related software installed?  
  **No**

gudvinr, 2023-01-21:
So I did some `git bisect` and found out that issue started to happen (if it is indeed caused by qmk) before 0.14.3 because 0.14.3 produces this error.

However, it's impossible to build image with `util/docker_build.sh` because container produces an error about missing python packages and docker images in registry are not tagged properly.

gudvinr, 2023-01-22:
I finally got some time to debug this, built `qmkfm/base_container` at `9fc31b63ac2ef521853438aaba577852cf4701af` and found out through bisect that c66df1664497546f32662409778731143e45a552 is the first bad commit. Thus 0.10.54 is the latest working release.

@noroadsleft it looks like your commit and seems rather large

noroadsleft, 2023-01-22:
I have a commitment this morning, but I should be able to investigate this in the afternoon. (It's 8:21 AM local time right now.)

gudvinr, 2023-01-22:
I mean, I can wait a bit more since I was able to cope with that for more than a year, lol. It's not an urgent matter at all.

gudvinr, 2023-01-22:
So, I did some more digging, since this commit looks like it was merged squashed from PR #11053, I did bisect from 0.10.54 to pre-develop-merge-nov20 and pinpoint first bad commit there:
75a18e69f9d3b6dfa470d0a7dbd78408d6a1c496 is the first bad commit

drashna, 2023-01-23:
Considering that the atomic block commit may be the issue, does adding `#define IGNORE_ATOMIC_BLOCK` to your config.h fix the issue?

gudvinr, 2023-01-23:
Yes, with this workaround issue I no longer see the issue but I don't think that this is a fix if users need to add this themselves for each board. This bug is rather obscure so it is kinda tricky to figure this out.  
There should be more general approach.