qmk/qmk_firmware #15428: QMK Compile not finding folder

JoakoV3, 2021-12-07:
I have tried to compile a keyboard with qmk MSYS and when I enter `qmk compile -kb planck -km default` It returns
```
qmk compile -kb planck -km default
usage: qmk.exe compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP] [-kb KEYBOARD] [filename]
qmk.exe compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'planck'

```

I tried putting absolute paths to it and all kinds of keyboard_folder values but it is just not working. I tried changing my path to keyboards, planck, qmk_firmware, but it's the same output on all paths.

Is there a solution to this?

fauxpark, 2021-12-07:
There is no keyboard in the repo named `planck`.
[image]

JoakoV3, 2021-12-07:
Oh okay, but when I put the `qmk compile -kb planck/rev1 -km default` it gives me an error 

```
tmk_core/protocol/lufa.mk:13: lib/lufa/LUFA/makefile: No such file or directory 
make: *** [Makefile:478: planck/rev1:default] Error 1
```
and says Make finished with errors.

fauxpark, 2021-12-07:
https://docs.qmk.fm/#/newbs

JoakoV3, 2021-12-07:
weird bug, had to run qmk setup again. closing...

closed 2021-12-07 by JoakoV3