qmk/qmk_firmware #15434: compile_commands.json

nathanvercaemert, 2021-12-08:
Makefile doesn't lend itself to generating compile_commands.json with bear. This would be to use a language server (ccls) with the repository. Is there a workaround for this? I dug up a post on Reddit, but it's a few years old and isn't quite perfect anymore.

tzarc, 2021-12-08:
`qmk generate-compilation-database -kb <yourkb> -km <yourkeymap>` should suffice, nowadays.

nathanvercaemert, 2021-12-08:
That's what I was looking for. Thanks for this. Do you know if it's referenced anywhere?

tzarc, 2021-12-08:
Yep, just not in an obvious position by the looks of it:

https://docs.qmk.fm/#/cli_commands?id=qmk-generate-compilation-database

nathanvercaemert, 2021-12-08:
Thanks. This will be helpful if I ever have to piece it together in the future. Thanks again for the help.

nathanvercaemert, 2021-12-08:
'''
➜ qmk_firmware (master) ✗ qmk generate-compilation-database -kb ergodox_ez -km nathanvercaemert
usage: qmk [-h] [-V] [-v] [--datetime-fmt DATETIME_FMT]
           [--log-fmt LOG_FMT] [--log-file-fmt LOG_FILE_FMT]
           [--log-file-level {debug,info,warning,error,critical}]
           [--log-file LOG_FILE] [--color] [--no-color]
           [--unicode] [--no-unicode] [--interactive]
           [--config-file CONFIG_FILE]
           {config,clone,console,env,setup,c2json,cd,chibios-confmigrate,clean,compile,doctor,flash,generate-rgb-breathe-table,info,json2c,lint,list-keyboards,list-keymaps,list-layouts,new-keyboard,new-keymap}
           ...
qmk: error: argument {config,clone,console,env,setup,c2json,cd,chibios-confmigrate,clean,compile,doctor,flash,generate-rgb-breathe-table,info,json2c,lint,list-keyboards,list-keymaps,list-layouts,new-keyboard,new-keymap}: invalid choice: 'generate-compilation-database' (choose from 'config', 'clone', 'console', 'env', 'setup', 'bux', 'c2json', 'cd', 'cformat', 'chibios-confmigrate', 'clean', 'compile', 'docs', 'doctor', 'fileformat', 'flash', 'format-c', 'format-json', 'format-python', 'format-text', 'generate-api', 'generate-config-h', 'generate-dfu-header', 'generate-docs', 'generate-info-json', 'generate-keyboard-h', 'generate-layouts', 'generate-rgb-breathe-table', 'generate-rules-mk', 'generate-version-h', 'hello', 'info', 'json2c', 'lint', 'list-keyboards', 'list-keymaps', 'list-layouts', 'kle2json', 'multibuild', 'new-keyboard', 'new-keymap', 'pyformat', 'pytest')
'''

Any suggestions?

tzarc, 2021-12-08:
Sounds like your repo is out of date, it was added to the repo on November 27th, this year.

nathanvercaemert, 2021-12-08:
I resolved this issue with help from https://github.com/qmk/qmk_firmware/issues/4855

I just needed to run:
```
bear -vvvv qmk compile
```

nathanvercaemert, 2021-12-08:
O wow, love how QMK is always being improved. Thanks for the help. Closing issue.

closed 2021-12-08 by nathanvercaemert