qmk/qmk_firmware #15442: Does the encoder and layer indicator support it?

jeromeli002, 2021-12-09:
Does it only support EC11 encoder? The mouse encoder is different from the conventional encoder. If it is spaced in the middle, it will turn to some position and it will always trigger. How to change it? Is there a reference keyboard code? Can the indicator light show the layer? For example, when working on layer 1, LED1 will light up, when working on layer 2, LED2 will light up …………

drashna, 2021-12-24:
Do you mean for the ploopy devices?

jeromeli002, 2021-12-25:
> 你是说那些笨拙的设备吗？

Like a mouse encoder

drashna, 2021-12-25:
The current encoder code only supports rotary encoders, yes.  It doesnt track the position but rather the changes produced by turning it. 

Essentially, it is doing this:

https://en.wikipedia.org/wiki/Rotary_encoder#/media/File:Incremental_directional_encoder.gif

jeromeli002, 2021-12-25:
> 当前编码器代码仅支持旋转编码器，是的。它不跟踪位置，而是跟踪转动它产生的变化。
> 
> 本质上，它是这样做的：
> 
> https://en.wikipedia.org/wiki/Rotary_encoder#/media/File:Incremental_directional_encoder.gif

https://item.szlcsc.com/476834.html
Do you want to modify the driver? The desktop should be changed to allow him to support this kind of encoder, and the kind of encoder on the mouse

peepeetee, 2021-12-29:
I just corresponded with @jeromeli002 and the question is as follows:
They've got a 30 Detent 15 PPR encoder, and is trying to get it to work with QMK. How would they do that?

closed 2022-06-18 by github-actions[bot]