qmk/qmk_firmware #15443: [Translation]Translation progress for zh-cn / 简体中文翻译进度

peepeetee, 2021-12-09:
Here's an issue to track the progress of translations of the docs to zh-cn, please ask about it/ post progress here.

此 issue 将追踪简体中文的翻译进度，请在这里问/交流翻译

peepeetee, 2021-12-09:
> Here's an issue to track the progress of translations of the docs to zh-cn, please ask about it/ post progress here.
> 
> 此 issue 将追踪简体中文的翻译进度，请在这里问有关问题/交流翻译

peepeetee, 2021-12-09:
@epiciskandar

peepeetee, 2021-12-09:
#14924 contains all changes I proposed here, and it looks good to go in already
#6924 should be paid more attention to since it did not stop me from doing duplicate work

peepeetee, 2021-12-13:
This now have a complete list of files translated in #14924

peepeetee, 2021-12-14:
I am planning to split up #14924 into manageable chunks, possibly single files. I shall track the progress here

epiciskandar, 2021-12-21:
github's dark theme definitly unfriendly with checkboxes, too hard to recongnize🙁

peepeetee, 2022-01-02:
need to catch up with #15678

givedrug, 2022-01-02:
hello！可以加入你们吗，想参与一起翻译~
contact me via givedrug@outlook.com

epiciskandar, 2022-01-03:
follow up: #15716

epiciskandar, 2022-01-03:
follow up: #15719

epiciskandar, 2022-01-12:
follow up #15856