qmk/qmk_firmware #15452: [Bug] ploopyco trackball mini axis inverted

strikersh, 2021-12-10:
## Describe the Bug

Got my ploopyco trackball mini a few days ago, and decided to customize it, after building and flashing the default layout axis were inverted, all the layout I tried a custom one included had the same issue. Rolling the ball to the right made the cursor go left, rolling up went down, etc.

I had to reflash the firmware provided on [ploopyco github](https://github.com/ploopyco/mini-trackball/blob/main/ploopyco_trackball_mini_rev1_002_default.hex), to restore the default behavior. That's why I think it's a regbression

## System Information

 - Keyboard:  ploopyco trackball mini 
   - Revision (if applicable):  rev1_002
 - Operating system: gentoo and archlinux
 - AVR GCC version: 11.2.1, but the same behavior happen with the docker image and qmk configurator
 - QMK Firmware version: 0.15.5
 - commit : bff5807b5b63011c954ba9a27c94eedf31352fa3

strikersh, 2021-12-10:
I tried defining `POINTING_DEVICE_INVERT_X`, `POINTING_DEVICE_INVERT_Y`

but it seams that the issue lie somewhere else that the axis assignation

drashna, 2021-12-13:
Try addin `#undef POINTING_DEVICE_ROTATION_90`

drashna, 2021-12-13:
If that makes it worse, then add `#define POINTING_DEVICE_ROTATION_270`

strikersh, 2021-12-13:
Just tested that didn't change anything