qmk/qmk_firmware #15453: [Bug] Hex value typo in quantum/rgblight_list.h causes

Shelleloch, 2021-12-10:
`rgblight_list.h` Line 32 is currently `#define RGB_AZURE 0x99, 0xf5, 0xFF`. The lowercase `f` causes a compiler failure; capitalising it fixes this.

Intended line:
`#define RGB_AZURE 0x99, 0xF5, 0xFF`

tzarc, 2021-12-11:
Please list the compiler failure -- 0xf5 is a valid hexadecimal value as far as the compiler is concerned.

Shelleloch, 2021-12-11:
```
Compiling: keyboards/ymdk/ymd09/ymd09.c                                                            In file included from quantum/rgblight.h:181,
                 from quantum/quantum.h:32,
                 from keyboards/ymdk/ymd09/ymd09.h:3,
                 from keyboards/ymdk/ymd09/ymd09.c:1:
quantum/rgblight_list.h:32: error: "RGB_AZURE" redefined [-Werror]
 #define RGB_AZURE 0x99, 0xf5, 0xFF

In file included from drivers/ws2812.h:18,
                 from quantum/rgblight.h:179,
                 from quantum/quantum.h:32,
                 from keyboards/ymdk/ymd09/ymd09.h:3,
                 from keyboards/ymdk/ymd09/ymd09.c:1:
./quantum/color.h:27: note: this is the location of the previous definition
 #define RGB_AZURE       0x99, 0xF5, 0xFF

cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:457: .build/obj_ymdk_ymd09_soundboard/keyboards/ymdk/ymd09/ymd09.o] Error 1
Make finished with errors
make: *** [Makefile:478: ymdk/ymd09:soundboard] Error 1```
Went back and changed it to cause the fail.

drashna, 2021-12-11:
the location of rgblight_list.h was moved, and the actual rgb defines moved to color.h. 

If you improperly updated, it's possible that you have both, and that would cause issues. 

Also, because they aren't identical, that's why it's creating a compiler error.

zvecr, 2021-12-12:
Removed label/renamed ticket as its not actually a bug on QMKs side of things.

Also, next time when filling out a bug report, you should fill in the template.

closed 2022-06-18 by github-actions[bot]