qmk/qmk_firmware #15497: [Feature Request] Support static brightness for RGB Matrix animations

ryanskidmore, 2021-12-17:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behaviour

## Description

Would it be possible to add an additional configuration option to statically define the brightness of the colours used in RGB matrix animations? Essentially the equivalent of statically setting the value component of HSV.

## Why?

I've just built a custom keyboard with per-key RGB, but unfortunately the switches I've chosen don't let through too much light. This means that the RGB animations look dim and can barely be seen behind the keys. I'm overriding the RGB LEDs for indicator keys and setting them to `RGB(255, 255, 255)` and they look great. I'm currently sitting at 190mA/500mA, so there's a fair bit of room to increase the brightness before I hit a power limit. Changing `RGB_MATRIX_MAXIMUM_BRIGHTNESS` seems to have no effect.

daskygit, 2021-12-17:
Have you tried increasing the brightness using the key code available after? The brightness gets stored in eeprom.

ryanskidmore, 2021-12-17:
@daskygit That worked - thank you for the pointer!

closed 2021-12-17 by ryanskidmore