qmk/qmk_firmware #15502: Questions about timer units

Weirdo-F, 2021-12-17:
When I use the timer_read() and timer_elapsed functions, I get results in ms, but I need either ns or cycle precision. How can I do that?

Weirdo-F, 2021-12-17:
Maybe there's another function or method to implement timers in ns or cycles? @drashna

drashna, 2021-12-17:
Not really, unfortunately.  

There may be chibiOS specific calls that can be used for ARM, but I'm not sure about AVR.

Weirdo-F, 2021-12-17:
> 不幸的是，不是真的。
> 
> 可能有一些特定于chibiOS的调用可用于ARM，但我不确定AVR。

> Not really, unfortunately.
> 
> There may be chibiOS specific calls that can be used for ARM, but I'm not sure about AVR.

I'm using STM32

Weirdo-F, 2021-12-17:
> Not really, unfortunately.
> 
> There may be chibiOS specific calls that can be used for ARM, but I'm not sure about AVR.

I'm using STM32

tzarc, 2021-12-17:
Step back. Why? What needs this sort of timing resolution?

Weirdo-F, 2021-12-18:
> 

I want to obtain the time of capacitor charging process, so as to determine whether the capacitance changes, that is, to realize the electrostatic capacitance keyboard.

Weirdo-F, 2021-12-18:
> Step back. Why? What needs this sort of timing resolution?

I want to obtain the time of capacitor charging process, so as to determine whether the capacitance changes, that is, to realize the electrostatic capacitance keyboard.

closed 2022-06-18 by github-actions[bot]