qmk/qmk_firmware #15503: [Feature Request] [Discussion] Update Auto Shift to Support Other Modifiers

defiant00, 2021-12-17:
I wanted an easy way to do _modifier + key_ on a hold, where the modifier could be specified as _shift_, _ctrl_, _alt_, or _gui_ (instead of being hard-coded as _shift_). I did a first pass on this, basing it on Auto Shift, which worked fine, but since this could be done a few different ways I wanted to discuss it here before submitting a pull request.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Here are the ways I can think of to implement this, along with pros and cons:

### 1. Do this as a simple update to Auto Shift

Add a new configuration flag for `rules.mk`:

    AUTO_SHIFT = shift | ctrl | alt | gui

And then in Auto Shift itself:
```c
#if defined(AUTO_SHIFT_CTRL)
    #define AUTO_SHIFT_KEY_L KC_LCTL
    #define AUTO_SHIFT_KEY_R KC_RCTL
#elif defined(AUTO_SHIFT_ALT)
    #define AUTO_SHIFT_KEY_L KC_LALT
    #define AUTO_SHIFT_KEY_R KC_RALT
#elif defined(AUTO_SHIFT_GUI)
    #define AUTO_SHIFT_KEY_L KC_LGUI
    #define AUTO_SHIFT_KEY_R KC_RGUI
#else
    #define AUTO_SHIFT_KEY_L KC_LSFT
    #define AUTO_SHIFT_KEY_R KC_RSFT
#endif
```
And then replace the hard-coded shift keys with `AUTO_SHIFT_KEY_L` and `AUTO_SHIFT_KEY_R`.

* Pros
  * No breaking changes
  * _Shift_ is the default behavior, so no need to update your keymap unless you want a different modifier
  * Very simple change that should just work, and firmware size should be identical
  * Single copy of the code, so easier to maintain
* Cons
  * The names don't really match functionality, which might be confusing and limit discoverability

### 2. Update and Rename

Do the changes outlined above for 1, and also go through and rename all the `#define`s, functions, etc. from `Auto Shift` to `Auto Mod`.

* Pros
  * Names and functionality match
  * Single copy of the code
  * Only minor changes to other core files to update names
* Cons
  * Breaking change
  * Would need to go through and fix existing keyboards

### 3. Create Auto Mod and leave Auto Shift as well

Create Auto Mod based off of Auto Shift, similar to 2 above, but keep both around for now.

* Pros
  * No breaking changes
  * Names match functionality
* Cons
  * Two copies of basically the same code to maintain
  * Two ways to do auto shift
  * Other core files will need to be updated to support Auto Mod as well

### Recommendation

I feel like the second approach is the best: having a single copy of the code, the names matching the functionality, and taking the time to fix existing keymaps.

With that said, please let me know what you think!

drashna, 2021-12-23:
for reference, Discord message thread: 
<https://discord.com/channels/440868230475677696/867530715192885269/923397345960411136>

IsaacElenbaas, 2022-01-11:
Both for individual keys and/or an entire board this can probably be 90% done by just replacing `KC_LSFT` in `autoshift_press/release_user` (see [here](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_auto_shift.md#custom-shifted-values)). Aside from that, the only things I can think of that would be an issue is `autoshift_flush_shift`, the oneshot key handling, and the physical shift state storing and cancelling under `autoshift_press`.

I don't really see the point, though. For individual keys Tap Dance is a much better approach, and before the recent PRs for keyrepeat and such I would have argued that Auto Shift should be rewritten on top of Tap Dance. None of the new features would really apply with other modifiers due to them not being used for typing.

closed 2022-06-18 by github-actions[bot]