qmk/qmk_firmware #15505: [Bug] KC_CAPS being sent as ctrl

Adam13531, 2021-12-18:
## Describe the Bug

Putting a `KC_CAPS` key in my Corne keymap results in ctrl being sent to the OS.

## System Information

 - Keyboard:  Corne
   - Revision (if applicable): [choc-v3](https://github.com/Adam13531/crkbd/blob/choc-v3/README.md#corne-keyboard)
 - Operating system: Windows 10
 - AVR GCC version: `8.4.0`
 - ARM GCC version: `10.1.0`
 - QMK Firmware version: `0.15.9`
 - Any keyboard related software installed? 
   - [ ] AutoHotkey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

My keymap is [here](https://github.com/Adam13531/qmk_firmware/tree/master/keyboards/crkbd/keymaps/adam). I could only repro this issue when the _left_ half of my Corne was plugged in via USB, but then it repros 100% of the time. It doesn't seem to matter which layer `KC_CAPS` is in, nor does it matter which half of the keyboard the key is on. It only matters which half was connected via USB.

To work around this, I had to change `KC_CAPS` to be a dual-function key (e.g. `LT(_FUN, KC_CAPS)`).

filterpaper, 2021-12-18:
It sounds like the [magic keycode](https://docs.qmk.fm/#/keycodes_magic?id=magic-keycodes) to swap Caps Lock and Ctrl may have been triggered. You can try:
- Clearing the EEPROM for both your controllers
- Or using the `CL_CAPS` keycode to un-swap it

Adam13531, 2021-12-19:
@filterpaper to the rescue—clearing the EEPROM on the left microcontroller did it! I had tried `CL_CAPS` before that, but it didn't seem to have any effect.

Thank you for the help!

closed 2021-12-19 by Adam13531