qmk/qmk_firmware #15521: Flashing Kint36 causes solid LEDs and keyboard is locked up

zemackdaddy, 2021-12-20:
I discovered this after fetching and merging every thing that is current into my local repo.

Previous to 84ea77ead663624a6d8ae451ca3d4318008d9d41, I was able to compile and flash my Kint36 (Teensy 3.6 for Kinesis Advantage).  Everything worked.

I believe this is related to:

* make the full 4096 bytes of EEPROM work on Teensy 3.6 (#12947)

If I use the commit above more recent, I can compile and flash but all of my LEDs are solid blue and the keyboard is locked up.

I am able to reset and flash just fine with the commit at or previous to 928d9f90ec169f49f50c1566132b06adcaf662d7.

I really don't know anything else about this or how it works. I just searched GitHub history for any commits related to this firmware and went backwards until I found the one that began working again.

Please let me know what else you want from me. Again, I can compile and flash just fine. It just breaks the keyboard with solid LEDs. I receive no errors during compilation or flashing.

zemackdaddy, 2021-12-20:
I've held in the reset button on my Teensy 3.6 for 17+ seconds and then refreshed.  No success.  Still flashes but is all solid LEDs.  If I flash back to 928d9f9, everything works fine.

stapelberg, 2021-12-21:
Maybe we need to revert that commit again after all and just live without flash until someone can properly fix it.

I have no idea why it works for some people but not for others.

zemackdaddy, 2021-12-24:
@stapelberg I am happy to help in anyway possible. I don't really understand the inner workings of the  code or the boards but I could enable debug on my side or something to help gather information.

zemackdaddy, 2021-12-29:
@stapelberg I got around to reverting the  commit in question and indeed I am able to flash and have the keyboard work again.

badgersow, 2022-01-01:
Just wanted to confirm that I'm experiencing exactly the same problem

stapelberg, 2022-01-01:
I have sent https://github.com/qmk/qmk_firmware/pull/15695 to revert the change for now.

The next step is figuring out which Teensy 3.6s are affected, i.e. why some work and others apparently don’t. Maybe a post in the PJRC forums would be a good idea. If anyone could help with this, that’d be appreciated. I don’t have time for it currently.

closed 2022-06-18 by github-actions[bot]