qmk/qmk_firmware #15587: The win key cannot be set, the bug is still a firmware compilation problem

jeromeli002, 2021-12-27:
The win key cannot be set. After any key is set to win, the key is invalid, and it is normal to set it to other keys. Is it a bug or a firmware compilation problem? The mcu is 32U4.

filterpaper, 2021-12-27:
Is "win key" referring to `KC_LWIN` and `KC_RWIN`?

jeromeli002, 2021-12-27:
> Is "win key" referring to `KC_LWIN` and `KC_RWIN`?

All useless, set to other keys is normal. Eliminate hardware problems

filterpaper, 2021-12-27:
They are aliases to the GUI key. Have you tried `KC_LGUI`?

peepeetee, 2021-12-27:
I see that you are using a translation program, do you need language help?

您看起来在用翻译工具，您需要语言帮助吗

jeromeli002, 2021-12-27:
> I see that you are using a translation program, do you need language help?
> 
> 您看起来在用翻译工具，您需要语言帮助吗

Google翻译的 ，就是编译的时候是设置的win键刷上后按键没反应 通过via设置也是一样，但是设置其他键是正常的排除硬件问题不触发
[image]

jeromeli002, 2021-12-27:
> They are aliases to the GUI key. Have you tried `KC_LGUI`?

[image]

peepeetee, 2021-12-27:
您看看是不是您电脑上有锁定windows键的软件？ 如果是笔记本的话试一下fn+windows

See if there's any software on your computer that affects the windows key

jeromeli002, 2021-12-27:
> 您看看是不是您电脑上有锁定windows键的软件？ 如果是笔记本的话试一下fn+windows
> 
> See if there's any software on your computer that affects the windows key

这样？ 笔记本自带的键盘win键是有用的能触发，晚上回去再试试吧

closed 2022-06-18 by github-actions[bot]