qmk/qmk_firmware #15596: [Bug] Pressing programmable button cause keyboard to not function anymore

Luc1412, 2021-12-27:
## Describe the Bug

I defined in my `kezmap.c` file in the default layout `PB_1`, `PB_2`, `PB_3` and `PB_4`. I also enabled programmable buttons in the `rules.mk` with `PROGRAMMABLE_BUTTON_ENABLE = yes`.
When I press any of the programmable buttons, the keyboard doesn't accept new inputs anymore until I reconnect it.

Here my fork with my current code: https://github.com/Luc1412/qmk_firmware/tree/master/keyboards/gmmk/pro/ansi/keymaps/luc1412

## System Information

 - Keyboard:  Glorious GMMK Pro
 - Operating system: Windows 11 Pro (21H2 build 22000.376)
 - AVR GCC version: 8.4.0
 - ARM GCC version: 10.1.0
 - QMK Firmware version: 0.15.9
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

knovoselic, 2021-12-30:
I have the same issue on Preonic Rev3

t-8ch, 2022-01-07:
Can you try with the following patch applied, it should work for both of you.

```patch
commit d2f5492fbd266a419bd02cd260a4d6a1ba01fdd8
Author: Thomas Weißschuh <thomas@t-8ch.de>
Date:   Fri Jan 7 12:04:09 2022 +0100

    ChibiOS: add support for HID Programmable Buttons

diff --git a/tmk_core/protocol/chibios/chibios.c b/tmk_core/protocol/chibios/chibios.c
index 41752b2130..b6cff05e9f 100644
--- a/tmk_core/protocol/chibios/chibios.c
+++ b/tmk_core/protocol/chibios/chibios.c
@@ -60,10 +60,11 @@ void    send_keyboard(report_keyboard_t *report);
 void    send_mouse(report_mouse_t *report);
 void    send_system(uint16_t data);
 void    send_consumer(uint16_t data);
+void    send_programmable_button(uint32_t data);
 void    send_digitizer(report_digitizer_t *report);
 
 /* host struct */
-host_driver_t chibios_driver = {keyboard_leds, send_keyboard, send_mouse, send_system, send_consumer};
+host_driver_t chibios_driver = {keyboard_leds, send_keyboard, send_mouse, send_system, send_consumer, send_programmable_button};
 
 #ifdef VIRTSER_ENABLE
 void virtser_task(void);
diff --git a/tmk_core/protocol/chibios/usb_main.c b/tmk_core/protocol/chibios/usb_main.c
index ac46a9312b..2106f5cb44 100644
--- a/tmk_core/protocol/chibios/usb_main.c
+++ b/tmk_core/protocol/chibios/usb_main.c
@@ -975,6 +975,35 @@ void send_consumer(uint16_t data) {
 #endif
 }
 
+void send_programmable_button(uint32_t data) {
+#ifdef PROGRAMMABLE_BUTTON_ENABLE
+    osalSysLock();
+    if (usbGetDriverStateI(&USB_DRIVER) != USB_ACTIVE) {
+        osalSysUnlock();
+        return;
+    }
+
+    if (usbGetTransmitStatusI(&USB_DRIVER, SHARED_IN_EPNUM)) {
+        /* Need to either suspend, or loop and call unlock/lock during
+         * every iteration - otherwise the system will remain locked,
+         * no interrupts served, so USB not going through as well.
+         * Note: for suspend, need USB_USE_WAIT == TRUE in halconf.h */
+        if (osalThreadSuspendTimeoutS(&(&USB_DRIVER)->epc[SHARED_IN_EPNUM]->in_state->thread, TIME_MS2I(10)) == MSG_TIMEOUT) {
+            osalSysUnlock();
+            return;
+        }
+    }
+    static report_programmable_button_t report = {
+        .report_id = REPORT_ID_PROGRAMMABLE_BUTTON,
+    };
+
+    report.usage = data;
+
+    usbStartTransmitI(&USB_DRIVER, SHARED_IN_EPNUM, (uint8_t *)&report, sizeof(report));
+    osalSysUnlock();
+#endif
+}
+
 void send_digitizer(report_digitizer_t *report) {
 #ifdef DIGITIZER_ENABLE
 #    ifdef DIGITIZER_SHARED_EP
```

knovoselic, 2022-01-08:
@t-8ch thanks for quick patch! I can confirm that the keyboard doesn't crash/freeze anymore, but I'm not sure if it's working properly as my system doesn't detect the keys. This is my first time using them, so I'm not sure exactly what I need to set up on the OS side. I'm running Manjaro with Gnome and Kernel 5.15, so it should be supported by the kernel. Any suggestions?

t-8ch, 2022-01-08:
The keys should be reported by `sudo libinput debug-events --show-keycodes` as `KEY_MACRO#`.

t-8ch, 2022-01-08:
Please let me know if the keys are received, then I will submit a PR with the patch.

knovoselic, 2022-01-08:
@t-8ch woohoo, it works. Thanks for the patch and help. Here's an example output:
```
-event21  KEYBOARD_KEY            +4.376s       KEY_MACRO1 (656) pressed
 event21  KEYBOARD_KEY            +4.448s       KEY_MACRO1 (656) released
 event21  KEYBOARD_KEY            +5.848s       KEY_MACRO2 (657) pressed
 event21  KEYBOARD_KEY            +5.912s       KEY_MACRO2 (657) released
 event21  KEYBOARD_KEY            +6.288s       KEY_MACRO3 (658) pressed
 event21  KEYBOARD_KEY            +6.336s       KEY_MACRO3 (658) released
 event21  KEYBOARD_KEY            +6.688s       KEY_MACRO4 (659) pressed
 event21  KEYBOARD_KEY            +6.752s       KEY_MACRO4 (659) released
 event21  KEYBOARD_KEY            +7.072s       KEY_MACRO5 (660) pressed
 event21  KEYBOARD_KEY            +7.128s       KEY_MACRO5 (660) released
 event21  KEYBOARD_KEY            +7.472s       KEY_MACRO6 (661) pressed
 event21  KEYBOARD_KEY            +7.536s       KEY_MACRO6 (661) released
 event21  KEYBOARD_KEY            +7.832s       KEY_MACRO7 (662) pressed
 event21  KEYBOARD_KEY            +7.920s       KEY_MACRO7 (662) released
 event21  KEYBOARD_KEY            +8.264s       KEY_MACRO8 (663) pressed
 event21  KEYBOARD_KEY            +8.336s       KEY_MACRO8 (663) released
 event21  KEYBOARD_KEY            +8.640s       KEY_MACRO9 (664) pressed
 event21  KEYBOARD_KEY            +8.704s       KEY_MACRO9 (664) released
 event21  KEYBOARD_KEY            +10.144s      KEY_MACRO10 (665) pressed
 event21  KEYBOARD_KEY            +10.192s      KEY_MACRO10 (665) released
 event21  KEYBOARD_KEY            +10.584s      KEY_MACRO11 (666) pressed
 event21  KEYBOARD_KEY            +10.624s      KEY_MACRO11 (666) released
```

commit f1cd2a5a (2022-02-04): ChibiOS: add support for HID Programmable Buttons (#15787)

* ChibiOS: add support for HID Programmable Buttons

Fixes #15596

* Enable SHARED_ENDPOINT when PROGRAMMABLE_BUTTON is enabled

The Programmable Button driver expects the shared EP to be enabled.
So enforce this invariant.
closed 2022-02-26 by zvecr