qmk/qmk_firmware #15598: [Bug] Flashing QMK firmware results in a non-working keyboard on Preonic rev3

lordwelch, 2021-12-27:
## Describe the Bug

Flashing QMK firmware results in a non-working keyboard.
After flashing the keyboard will go into one of two states:

1. It stays in DFU mode not allowing any keyboard functions.
2. It will "hold down" the first column of keys and the 7th column this happens whether the PCB is in the case or not and also happens whether or not switches are installed in the PCB.

The only exception to this is the preonic firmware located here https://docs.qmk.fm/#/faq_build?id=i-just-flashed-my-keyboard-and-it-does-nothingkeypresses-dont-register-its-also-arm-rev6-planck-clueboard-60-hs60v2-etc-feb-2019
which allows the keyboard to be used normally without issue.

This issue appears to be a rare issue and is also described here by another user https://www.reddit.com/r/olkb/comments/m9aabr/multiple_key_stuck_after_flashing_preonic_using/

In both places in mentions resetting the EEPROM with the reset firmware holding down the top left key when plugging in results in all LEDs turning off implying that the EEPROM does get reset however this has no effect on subsequent flashes of QMK, with any other firmware there is no noticeable change in behavior when attempting a Bootmagic EEPROM reset.

Attached is an example firmware built Mon Dec 27 19:29:20 UTC 2021 on commit 4e077250d56f7e786af0cdef00f4b41d77e2b85c that stays in DFU mode.

[preonic_rev3_default.bin.zip](https://github.com/qmk/qmk_firmware/files/7781538/preonic_rev3_default.bin.zip)

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  3
 - Operating system: macOS Monterey 12.1, MacBook Pro using the M1 Max
 - AVR GCC version: avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0

 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.15.11

 - Any keyboard related software installed? NO
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Note: the same symptoms appear when compiling on Fedora 34.

sigprof, 2021-12-27:
Can you try the `preonic/rev3_drop` firmware instead of `preonic/rev3`? The `rev3_drop` firmware uses an older version of custom matrix scanning code, and for some reason that version seems to work better on those strange boards where the normal version fails.

lordwelch, 2021-12-27:
Thanks that solved my problem. Looking through my command history I had started using `preonic/rev3_drop` but somehow dropped the `_drop`

closed 2021-12-27 by lordwelch