qmk/qmk_firmware #15600: [XAP] `info.json` transformation for XAP-related info

tzarc, 2021-12-27:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

**Any work in this area needs to be discussed with QMK Collaborators first, and in this case primarily with @tzarc due to their prior work in this area.**

The `info.json` needs to be "massaged" to only require relevant information for XAP. This includes enabled subsystems such as rgb_matrix and dynamic_keymap, as well as information like layout options.

Currently the `xap` branch of `qmk_firmware` compresses the entire output of `qmk info -f json` for a keyboard as a proof-of-concept; this is not sufficient as it currently does not include keymap-level selections (e.g. disabling rgb at the keymap is not reflected in the output), and the generated output also includes a lot of irrelevant information in the output that has no bearing on XAP's functionality.

For this issue, a transformation needs to be applied such that it filters for a set of known-used items from `info.json` into an appropriate equivalent to be embedded into the firmware.

A corresponding `*.jsonschema` file should also be provided so host applications can determine what information is to be available.

**NOTE:** for simplified porting from Chinese vendors who don't bother upstreaming to QMK, relevant information such as pin definitions should be included in this output. This is intended as a future work item; the idea is to be able to connect to a XAP-enabled firmware with the XAP client and export as much of a keyboard's `info.json` as possible.

tzarc, 2021-12-27:
This item also needs to add to the QMK subsystem in the XAP protocol, exposing a handler that allows retrieval of the `info.json`-like blob from the firmware.

This will need multiple XAP requests, and will likely need to be implemented by supplying a "read offset" as an argument to the handler; the host app requesting the information can then concatenate, decompress, and parse the final output.

tzarc, 2023-01-03:
Moved back to "To do" so that we can sort out the firmware-based jsonschema for clients.

zvecr, 2023-01-04:
Its currently just `data/schemas/keyboard.jsonschema`, do we expect to duplicate the content?