qmk/qmk_firmware #15633: 更新后编译报错 Compile error after update

jeromeli002, 2021-12-29:
[Administrator@bgs qmk_firmware]$ make zvecr/zv48/f401:default
QMK Firmware 0.15.11
Making zvecr/zv48/f401 with keymap default

arm-none-eabi-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: platforms/chibios/drivers/serial_usart.c                                                In file included from platforms/chibios/drivers/serial_usart.c:17:
platforms/chibios/drivers/serial_usart.h:21:10: fatal error: printf.h: No such file or directory
   21 | #include "printf.h"
      |          ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:457：.build/obj_zvecr_zv48_f401_default/serial_usart.o] 错误 1
Make finished with errors
make: *** [Makefile:478：zvecr/zv48/f401:default] 错误 1

peepeetee, 2021-12-29:
Try `make git-submodule` in qmk_firmware/

试试在 qmk_firmware/ 这里跑一下 `make git-submodule`

jeromeli002, 2021-12-29:
> 

[image]
还是一样的结果

peepeetee, 2021-12-29:
加个QQ？

jeromeli002, 2021-12-29:
> 加个QQ？

好了 打开文件夹发现只有一个 .git删了在跑一次就出来了

peepeetee, 2021-12-29:
您

peepeetee, 2021-12-29:
您好了吗？ 好了把 issue 关一下

peepeetee, 2021-12-29:
看您有好多 issue，加个QQ我来提供语言/翻译帮助？

jeromeli002, 2021-12-29:
> 看您有很多问题，加个QQ我来提供语言/翻译帮助？

532192607

closed 2021-12-29 by jeromeli002