qmk/qmk_firmware #15679: [Bug] With RGB_DISABLE_WHEN_USB_SUSPENDED lights still turn on when keys are pressed when computer is powered off

xard-dev, 2021-12-31:
When `RGB_DISABLE_WHEN_USB_SUSPENDED` is used RGB backlights turn on when keys are pressed when computer is powered off.

## Details

Using QMK on GMMK Pro ISO with additional  `RGB_DISABLE_WHEN_USB_SUSPENDED` and `RGB_DISABLE_TIMEOUT` configuration options the USB suspend works but has a minor issue: The RGB still turns on if keys are accidentally pressed when the computer is powered off.

When this happens constant backlight mode looks normal but with any animation mode the animations are advanced only if keys are pressed (which further indicates that this is not how the RGB is supposed to work). 

## Mitigation

Manually disable RGB or set `RGB_DISABLE_TIMEOUT` to low value which will eventually turn off the backlight automatically. 

## System Information

 - Keyboard:  GMMK Pro ISO
 - Operating system: Ubuntu 20.04
 - AVR GCC version: 5.4.0
 - ARM GCC version:  9.2.1
 - QMK Firmware version:  0.15.12
 - Any keyboard related software installed? 
   - None

## Additional Context

Stock GMMK Pro firmware does not power the RGB lights when keys are pressed while computer is turned off.

xard-dev, 2022-01-04:
I'm not so sure about the `RGB_DISABLE_TIMEOUT` mitigiation actually working as I tried it. My keyboard shuts the RGB as expected as long as my computer is powered on but not so much when powered off.

I looked for potential issue point from the `quantum/rgb_matrix/rgb_matrix.c` 391 around row assuming this is the right place:
```
    // Ideally we would also stop sending zeros to the LED driver PWM buffers
    // while suspended and just do a software shutdown. This is a cheap hack for now.
    bool suspend_backlight = suspend_state ||
#if RGB_DISABLE_TIMEOUT > 0
                             (rgb_anykey_timer > (uint32_t)RGB_DISABLE_TIMEOUT) ||
#endif  // RGB_DISABLE_TIMEOUT > 0
                             false;

    uint8_t effect = suspend_backlight || !rgb_matrix_config.enable ? 0 : rgb_matrix_config.mode;
```

Despite the comments I'd assume that even if `suspend_state` gets somehow turned off during suspend the `RGB_DISABLE_TIMEOUT` should act as backup unless the `rgb_anykey_timer` counting is not running. Also I'm assuming that `uint8_t effect` with value 1 equals RGB light turned off.

GeorgeGedox, 2022-02-05:
I'm also facing this bug, not using the `RGB_DISABLE_TIMEOUT` variable. At this point I just disable the RGB manually before turning off the computer.
Is this issue common for all keyboards using QMK or just an isolated bug?

airmack, 2022-02-06:
Can confirm with YMDK/wings. 
It is worse, if the keyboard is connect  to an external USB-hub. 
The keyboard/LEDs turns back on after around 5 seconds, if  it is connected to the external hub. 

I can't really reproduce this behavior, when the keyboard is directly connected to the PC...

drashna, 2022-03-10:
unfortunately, the suspend behavior here is determined by the USB state.   If it's coming back on, then it's likely be registered as a non-sleep state.   I'm not sure there is a simple fix for this.

xard-dev, 2022-03-11:
> unfortunately, the suspend behavior here is determined by the USB state. If it's coming back on, then it's likely be registered as a non-sleep state. I'm not sure there is a simple fix for this.

The suspend state still is enabled but it does not stay there which raises questions about the USB behavior itself. Still the stock closed firmware manages to handle this just fine with the very same hardware.

l33tlinuxh4x0r, 2022-10-13:
I have a Keychron Q6 and the stock firmware doesn't have this issue, however any firmware built from the qmk github source for this keyboard does in fact have the same issue. Sleep works but if you press a key when the computer is off the keyboard lights up. Only half of an RGB cycle appears to happen though and then it stops and freezes so half of the keyboard is lit and not doing any motion. Turning on the computer and everything works as normal.

xard-dev, 2022-10-16:
> I have a Keychron Q6 and the stock firmware doesn't have this issue, however any firmware built from the qmk github source for this keyboard does in fact have the same issue. Sleep works but if you [...]

I actually haven't tried flashing my own firmware to the Q series keyboards as everything I've so far wanted from the firmware works out of the box.

That raises some interesting questions like:
- What are the differences between stock Q firmware and self compiled one?
- Is the compilation environment related to this behavior?

l33tlinuxh4x0r, 2022-10-16:
> > I have a Keychron Q6 and the stock firmware doesn't have this issue, however any firmware built from the qmk github source for this keyboard does in fact have the same issue. Sleep works but if you press a key when the computer is off the keyboard lights up. Only half of an RGB cycle appears to happen though and then it stops and freezes so half of the keyboard is lit and not doing any motion. Turning on the computer and everything works as normal.
> 
> I actually haven't tried flashing my own firmware to the Q series keyboards as everything I've so far wanted from the firmware has worked out of the box.
> 
> That raises some interesting questions like:
> 
>     * What are the differences between stock Q firmware and self compiled one?
> 
>     * Is the compilation environment related to this behavior?

I was using sonixqmk (based on qmk) for a redragon keyboard and they also had issues with RGB suspend but they fixed it. IIRC they said something about chibios version and I think that they have a patch for newer versions but I couldn't find anything referencing it when I looked yesterday.

l33tlinuxh4x0r, 2023-01-08:
> I have a Keychron Q6 and the stock firmware doesn't have this issue, however any firmware built from the qmk github source for this keyboard does in fact have the same issue. Sleep works but if you [...]

For what it is worth the latest stock firmware has this issue. So something has changed in qmk.

chronocide-commando, 2023-01-22:
RGB appears to sometimes turn on from electrostatic discharge or vibration — without a keypress registering (I have "Power on by keyboard" enabled, and this doesn't power on the computer unlike when I actually press a key).
Strictly speaking, it happens during Standby (S3), since I hardly ever power the computer off completely, but it looks like the same bug.

geezmolycos, 2023-06-11:
You can disable waking up your keyboard with keypresses [by patching this function](https://github.com/qmk/qmk_firmware/blob/dd8b4dce55bd1845c69d4472bc3ce6338dcfa278/platforms/suspend.c#L43) to make the function return false without performing an actual check of keypresses.
Comment out those lines:
```c
bool suspend_wakeup_condition(void) {
    // COMMENT OUT FOLLOWING LINES
    /*matrix_power_up();
    matrix_scan();
    matrix_power_down();
    for (uint8_t r = 0; r < MATRIX_ROWS; r++) {
        if (matrix_get_row(r)) return true;
    }*/
    return false;
}
```
Note that this change will also disable waking up the computer with keyboard press.

xard-dev, 2023-06-14:
> You can disable waking up your keyboard with keypresses [by patching this function](https://github.com/qmk/qmk_firmware/blob/dd8b4dce55bd1845c69d4472bc3ce6338dcfa278/platforms/suspend.c#L43) to make [...]

While technically that should fix the issue it disabled waking up on sleep for laptops or desktops which is not good.

xard-dev, 2023-12-02:
Building a 0.22.9 release of QMK (while also remembering to update all the git sub modules) fixed this issue for me.

With the tested QMK build my GMMK Pro keyboards do turn the backlight off when a computer is turned off and the lights do not turn on if I touch the keys while the computer is off. Backlight is also turned off when the computer goes to sleep and waking up the computer by pressing any key works.

closed 2023-12-02 by xard-dev