qmk/qmk_firmware #15709: [QMK toolbox] Firmware missing/unable to download {Xbows keyboard series}

darknessblade, 2022-01-02:
I am unable to download any of the firmware versions trough the QMK Toolbox for the Xbows Keyboard series.
Other firmware's for keyboards like the Ergodox Do work when i try to download them.
[image]

Only if i go trough the QMK creator, am i able to download a Firmware version for the Keyboard.

I rather not to go trough the creator itself, as i want a 100% clean Keyboard layout/layer settings, that only contains the basic factory settings, of which i am uncertain that this is the case with the version for the QMK creator
----------------------
PS: i do not have my keyboard yet, [ordered it already, but have not received it yet], 

As I wanted to get Familiar with the QMK creator/toolbox before actually flashing my keyboard, i already downloaded the needed programs.

fauxpark, 2022-01-02:
The "keyboard form qmk.fm" section is deprecated and will be removed in the next release. The list is not being updated anymore due to Travis being shut down.

closed 2022-01-02 by fauxpark

darknessblade, 2022-02-06:
[fauxpark] 
Has there been any update regarding the QMK toolbox being updated so it gets the correct version/firmware.

Making it no longer uses the "Travis" page for its firmware

as the issue is still present in the current latest version.
----------
[received my keyboard recently, and wanted to look into enabling **VIA** support on the keyboard. so i do not have to re-flash the keyboard every time i add a new feature to one of the layers]
---------

fauxpark, 2022-02-06:
The next version will not have this functionality at all. The entire "keyboard from qmk.fm" section has already been removed in the current code because the mechanism that compiles and uploads the firmware no longer exists.

No new release has been made yet; I want to get to that soon.

darknessblade, 2022-02-07:
AAH OK, then i will wait for the next release before updating/upgrading the firmware for my keyboard. and Enable VIA support.