qmk/qmk_firmware #15710: QMK repeat rate and repeat delay

givedrug, 2022-01-02:
I'd like to change two parameters.
1、Repeat rate. If you hold down a key (eg. A), after a short time that key gets repeatedly sent (eg. the result is aaaaaaaaaaaaaaaaa). I'd like to change the rate at which the repeated output is sent (eg. to send more/less characters over a given time interval).
2、Repeat delay. How long you have to wait until a key is consired "held down" (eg. to go from 200 milliseconds to 60 milliseconds, or whatever).
How can I do this with the QMK/TMK firmware? I tried messing with #define TAPPING_TERM 200, but it didn't work.

i know It's not an option that can be set.
but can i use some custom handling/code to repeat keys at a particular rate.

t-8ch, 2022-01-06:
Normally this logic is all done by the operating system on the host side.
The keyboard only reports "key pressed" and "key not pressed".

You can also emulate this in QMK but that's not done by default.

See also:
https://teddit.net/r/olkb/comments/8e5c1c/qmk_repeat_rate_and_repeat_delay/
https://teddit.net/r/KeyboardLayouts/comments/pg85sh/repeat_last_key_in_qmk/
https://teddit.net/r/olkb/comments/hxmtkp/how_to_let_qmk_repeat_the_keypress/

givedrug, 2022-01-07:
> Normally this logic is all done by the operating system on the host side. The keyboard only reports "key pressed" and "key not pressed".
>
> You can also emulate this in QMK but that's not done by [...]

thx, i'll have a try!

KoenGoe, 2022-03-24:
> Normally this logic is all done by the operating system on the host side.

True for keycodes that actually go to the OS, but for instance for `RGB_VAI`it might be nice to have such functionality built in

closed 2022-07-24 by github-actions[bot]