qmk/qmk_firmware #15735: Request: Beta Docs Redirect

redxtech, 2022-01-04:
I've been learning how to use qmk recently, and I'm loving it. What I'm not loving is the way that every time I google how to do something with qmk, every single result is to the beta docs site, which isn't live anymore.

Is there any way to set up a redirect to the equivalent page in the live docs?

AffluentOwl, 2022-01-05:
Bruh, seriously, I was just about to open this issue. docs.qmk.fm hasn't been crawled by Google since 2017 no Google links other than the home page work. Likely because the docs are rendered entirely on the client side using javascript, and have hashbang mode URL's instead of HTML5 URL's.

What's the point of documentation if nobody can find it or search through it for the relevant section?

tzarc, 2022-01-05:
`beta.docs.qmk.fm` was out of date by more than 6 months, so wasn't relevant to QMK as of today anyway.
You can thank Travis CI and their gutting of open-source free execution time for that.

We have an alternative in the pipeline, but because of the scale of the docs it's taking some time. For now, sit tight.

closed 2022-06-18 by github-actions[bot]