qmk/qmk_firmware #15768: [Bug] Rotary encoders on my kyria starts sending two key presses after upgrading to 0.14

plattfot, 2022-01-07:
## Describe the Bug
I just pulled the latest changes from master and rebuild the firmware for my kyria. As I'm experimenting with enabling combos. But following the latest docs on how to enable that generated build errors. Which went away when updating to commit db8cf0b872. Now however when I use my rotary encoders they send out two key presses per click instead of one like they did before.

I have narrow it down to be somewhere between v0.13.38 and v0.14.0.

## System Information

 - Keyboard:  
   - Revision (if applicable):  db8cf0b872
 - Operating system: Arch Linux (5.15.12-arch1-1)
 - AVR GCC version: 11.2.0

 - ARM GCC version: 11.2.0

 - QMK Firmware version: 0.14.19

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I have tried changing the `ENCODER_RESOLUTION` from `2`, to `1` and `4` but that just made it worse.
I've also tried adding:

```
#define TAP_CODE_DELAY 10
```
To my `config.h` and still the same issue.

My keymap is here: [keyboards/splitkb/kyria/keymaps/plattfot](https://github.com/qmk/qmk_firmware/tree/master/keyboards/splitkb/kyria/keymaps/plattfot)

actarien, 2022-01-19:
I have the same problem, also with a Kyria keyboard ! But I've not yet found how to fix it (i'm new in qmk software)…

daskygit, 2022-01-19:
Looking at the the keymap linked above https://github.com/qmk/qmk_firmware/blob/91e50209e74f870d2195753402ab6816d1e378be/keyboards/splitkb/kyria/keymaps/plattfot/keymap.c#L399 should be changed to 
```c
return false;
```
otherwise you will also run the encoder function here
https://github.com/qmk/qmk_firmware/blob/master/keyboards/splitkb/kyria/rev1/rev1.c#L132-L154

plattfot, 2022-01-19:
That was it! Thank you. I'll need to read the docs more carefully next time.

Leaving this open, if this didn't solve it for @actarien.

actarien, 2022-01-19:
I had the same problem… and the solution worked also for me, many thanks !!
I think that before the splitkb/kyria keyboard reorganisation, returning 'true' didn't matter…

One thing is puzzling me : changing 'return true;' to 'return false;' in my keymap.c builds a new firmware shorter than before ?! Strange, but nice, as I'm around limit !

plattfot, 2022-01-19:
Oh, I didn't check the size when I recompiled. But I'm not that surprised it shrunk.  Since it doesn't need to call `encoder_update_kb` in the `rev1.c` code anymore, the linker can just through that away.

Closing this issue, now that it's been resolved for both of us.

closed 2022-01-19 by plattfot

Eloren1, 2024-11-03:
This method worked for me: