qmk/qmk_firmware #15773: [Feature Request] Supporting dip switches on both halves of a split keyboard.

juliusroeder, 2022-01-07:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Currently dip switches are only supported on the master side of a split keyboard. 
I'd like to add support for dip switches on both halves of a split keyboard in a similar fashion as for encoders. 
I have the enhancement ready in a branch on my fork and I'd be happy to make a PR.

Onefabis, 2022-06-04:
Seems like the feature already here, but it doesn't work, at least from my side: dip works only from the master, side. Slave doesn't work.

juliusroeder, 2022-08-28:
> Seems like the feature already here, but it doesn't work, at least from my side: dip works only from the master, side. Slave doesn't work.

As far as i remember it was only fully implemented on the master half. 
Never on the slave half... So there was also no actual sending and the master half wasn't ready for receiving the information either. 
Maybe that has changed. 

If you want I have a full implementation that I use. However, by now its far behind master...
I'd be happy to rebase it and make a pull request if there is interest from the project maintainers. 

https://github.com/juliusroeder/qmk_firmware/tree/feature/dip_switch_split_keyboard

closed 2022-12-28 by github-actions[bot]