qmk/qmk_firmware #15804: Debug Help With One Shot Keys

morgankenyon, 2022-01-09:
I am having trouble with one shot keys working on my new keyboard, and wanted to know how what the best way to track done what might be going on.

My keyboard is a crkbd, and here is my [keymap](https://github.com/morgankenyon/qmk_firmware/blob/master/keyboards/crkbd/keymaps/morgan.kenyon/keymap.c). On my keyboard I have both `OSM(MOD_LSFT),` and `OSM(MOD_LCTL)` keys defined. If I press both `OSM(MOD_LSFT)` and an alpha character at the same time it will print out uppercase. Same behavior with `OSM(MOD_LCTL)`. Press and hold always works, but 95% of the time its definitely not remaining active till the next key is pressed. Every once in awhile it will work as expected, but it happens very rarely.

I was able to turn debugging by following some debugging tips online and have added the following to my `keymap.c`, 

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  // If console is enabled, it will print the matrix position and status of each key pressed
#ifdef CONSOLE_ENABLE
    uprintf("KL: kc: %u, col: %u, row: %u, pressed: %u\n", keycode, record->event.key.col, record->event.key.row, record->event.pressed);
#endif 
  return true;
}
``` 

I receive the following logs when pressing each key.

`OSM(MOD_LSFT)`
```
KL: kc: 21762, col: 0, row: 1, pressed: 1
KL: kc: 21762, col: 0, row: 1, pressed: 0
```

`OSM(MOD_LCTL)`
```
KL: kc: 21761, col: 0, row: 2, pressed: 1
KL: kc: 21761, col: 0, row: 2, pressed: 0
```
 
It seems like because I am receiving both the `0` and `1` message that the "stickiness" of my one shot keys aren't holding.

Do you have any suggestions on how to further debug this issue? 

Thanks!

morgankenyon, 2022-01-09:
I am also on a windows 11 machine, and have defined the following two `one_shot` defines [here](https://github.com/morgankenyon/qmk_firmware/blob/master/keyboards/crkbd/keymaps/morgan.kenyon/config.h#L36-L37).

```c
#define ONESHOT_TAP_TOGGLE 5
#define ONESHOT_TIMEOUT 10000
```

If I do not define `ONESHOT_TIMEOUT`, will the one shot functionality continue indefinitely? That "requiredness" of that define was something the documentation didn't specify.

filterpaper, 2022-01-10:
`process_record_user` will be called when a key is pressed and when it is released. Hence the logging of `pressed: 0` is normal and not an indication of an issue.

One-shot mod keys will behave exactly like a normal mod when you press and hold it, followed by another. To use one shot, tap it once (don't hold) to register the mod, followed by another key to apply that mod to.

morgankenyon, 2022-01-10:
@filterpaper Thanks for the answer, sorry if I wasn't more clear, the "tap it once" (don't hold) part of the one shot keys is what is not working for me. I can only register the mod if I continue to hold the one shot keys.   

Do you know of any way to debug why qmk isn't applying a mod after tapping a one shot key? 

Since the key continues to work properly when held, my thought is its not a hardware issue, but I could be wrong about that.

morgankenyon, 2022-01-13:
@sigprof Thank you for letting me know the `#define TAPPING_FORCE_HOLD` breaks the toggle, I wasn't aware of that.

I saw some debugging help on the [one shot keys page](https://github.com/qmk/qmk_firmware/blob/master/docs/one_shot_keys.md), and included the following method in my [keymap.c](https://github.com/morgankenyon/qmk_firmware/blob/master/keyboards/crkbd/keymaps/morgan.kenyon/keymap.c#L86-L102). 

```c
void oneshot_mods_changed_user(uint8_t mods) {
  if (mods & MOD_MASK_SHIFT) {
    println("Oneshot mods SHIFT");
  }
  if (mods & MOD_MASK_CTRL) {
    println("Oneshot mods CTRL");
  }
  if (mods & MOD_MASK_ALT) {
    println("Oneshot mods ALT");
  }
  if (mods & MOD_MASK_GUI) {
    println("Oneshot mods GUI");
  }
  if (!mods) {
    println("Oneshot mods off");
  }
}
```

Here is the debug output after about 20 or so taps on the `OSM(MOD_LSFT)` key. 
```   
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
  > keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > Oneshot mods SHIFT
  > keyboard_report: 02 00 00 00 00 00 00 00 
    Oneshot mods off
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
  > keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > Oneshot mods SHIFT
  > keyboard_report: 02 00 00 00 00 00 00 00 
    Oneshot mods off
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
    keyboard_report: 00 00 00 00 00 00 00 00 
  > keyboard_report: 02 00 00 00 00 00 00 00 
  > keyboard_report: 00 00 00 00 00 00 00 00 
  > Oneshot mods SHIFT
  > keyboard_report: 02 00 00 00 00 00 00 00 
    Oneshot mods off
```

It's only registered the one shot shift three times. Would you have any idea why that behavior would be registered so infrequently, or if I could see more of what was going on with my keyboard?

morgankenyon, 2022-01-29:
I have been playing around with it a bit lately and still cannot accurately get it to work as expected. Here is a series of a's I typed attempting to use a one shot shift. So press the one shot shift and releasing, then pressing A. 

As you can see, more than half worked, but still not very reliable

```
AaaAAAAAAaAAaAaAaaaAAAAAAAAAAAAAAAAAAAaaAaAAAAA
```

morgankenyon, 2022-01-29:
I added the following to my keymap.c:

```c
void keyboard_post_init_user(void) {
  // Customise these values to desired behaviour
  debug_enable=true;
  //debug_matrix=true;
  //debug_keyboard=true;
  //debug_mouse=true;
}

void oneshot_mods_changed_user(uint8_t mods) {
  println("Oneshot mods");
  if (mods & MOD_MASK_SHIFT) {
    println("Oneshot mods SHIFT");
  }
  if (mods & MOD_MASK_CTRL) {
    println("Oneshot mods CTRL");
  }
  if (mods & MOD_MASK_ALT) {
    println("Oneshot mods ALT");
  }
  if (mods & MOD_MASK_GUI) {
    println("Oneshot mods GUI");
  }
  if (!mods) {
    println("Oneshot mods off");
  }
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
// If console is enabled, it will print the matrix position and status of each key pressed
#ifdef CONSOLE_ENABLE
    uprintf("KL: kc: %u, col: %u, row: %u, pressed: %u\n", keycode, record->event.key.col, record->event.key.row, record->event.pressed);
#endif 
  return true;
}
```

The frequency that this works dropped dramatically.

```
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaA
```

When typing a lowercase a, here are the logs I am receiving:

```
  > KL: kc: 21762, col: 0, row: 1, pressed: 1
  > KL: kc: 21762, col: 0, row: 1, pressed: 0
  > KL: kc: 4, col: 1, row: 1, pressed: 1
  > KL: kc: 4, col: 1, row: 1, pressed: 0
```
21762 = shift
4 = a

But the one it worked correctly these were the logs:

```

  > KL: kc: 21762, col: 0, row: 1, pressed: 1
    Oneshot mods
    Oneshot mods SHIFT
  > KL: kc: 21762, col: 0, row: 1, pressed: 0
  > KL: kc: 4, col: 1, row: 1, pressed: 1
    Oneshot mods
    Oneshot mods off
  > KL: kc: 4, col: 1, row: 1, pressed: 0
```

Is there an override method that allows me to see what qmk is doing when it decides whether a shift is a normal shift or a one shot shift? 

Just trying to dive deeper into why my one shot shift/ctrl aren't working.

sigprof, 2022-01-29:
Try also dumping `record->tap.count` and `record->tap.interrupted` (the tapping code need to detect the action as a tap for it to be interpreted as a one shot mod).

morgankenyon, 2022-02-20:
@sigprof That seems to be the underlying issue with my keyboard. The keyboard isn't registering my "tapping" of the key. Here are the results of me pressing "One shot shift + a" key.

When it isn't working, the Shift key is registered without "tap" being activated.
```
> KL: kc: 21762, pressed: 1, **tapped: 0**, inter: 0
  KL: kc: 21762, pressed: 0, tapped: 0, inter: 0
> KL: kc: 4, pressed: 1, tapped: 0, inter: 0
> KL: kc: 4, pressed: 0, tapped: 0, inter: 0
```

When it is working, the Shift key's tap is active.

```
> KL: kc: 21762, pressed: 1, **tapped: 1**, inter: 0
  One shot mods
  One shot mods SHIFT
  KL: kc: 21762, pressed: 0, tapped: 1, inter: 0
> KL: kc: 4, pressed: 1, tapped: 0, inter: 0
  One shot mods
  One shot mods off
> KL: kc: 4, pressed: 0, tapped: 0, inter: 0
```

Let me see if there is any way to maybe lengthen or shorten the time threshold needed in order to register a tap versus a regular press/release of that key.

Here are my debug functions.
```c
void oneshot_mods_changed_user(uint8_t mods) {
  println("Oneshot mods");
  if (mods & MOD_MASK_SHIFT) {
    println("Oneshot mods SHIFT");
  }
  if (mods & MOD_MASK_CTRL) {
    println("Oneshot mods CTRL");
  }
  if (mods & MOD_MASK_ALT) {
    println("Oneshot mods ALT");
  }
  if (mods & MOD_MASK_GUI) {
    println("Oneshot mods GUI");
  }
  if (!mods) {
    println("Oneshot mods off");
  }
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
// If console is enabled, it will print the matrix position and status of each key pressed
#ifdef CONSOLE_ENABLE
    uprintf("KL: kc: %u, pressed: %u, tapped: %u, inter: %u\n", keycode, record->event.pressed, record->tap.count, record->tap.interrupted);
#endif 
  return true;
}
```

morgankenyon, 2022-02-20:
Found this documentation that might hold some answers for me.

https://github.com/qmk/qmk_firmware/blob/master/docs/tap_hold.md

morgankenyon, 2022-02-20:
The `TAPPING_TERM` seems to be the root of my issue. It was initially set to 100 (`#define TAPPING_TERM 100`), but I bumped it up to 300 and it seems to be working great now. 

Thank you so much for your help @sigprof and @filterpaper!! 

I am going to close this issue out now.

closed 2022-02-20 by morgankenyon