qmk/qmk_firmware #15807: Continuing where someone left off for Koolertron no name 3x3 rgb kb

Jenpatrick, 2022-01-10:
Formatting update; this doesn't change any of the logic.

```suggestion
    [0] = LAYOUT(
        KC_NO,   BL_STEP,  KC_1,    KC_2,    KC_3,    KC_NO,   KC_MS_L, KC_MS_R,
        KC_ESC,  KC(L'ä'), KC_6,    KC_7,    KC_A,    KC_BSPC, KC_MS_U, KC_MS_D,
        KC_Q,    KC_W,     KC_E,    KC_H,    KC_I,    KC_J,    KC_WH_D, KC_WH_U,
        KC_A,    KC_S,     KC_D,    KC_P,    KC_UP,   KC_ENT,  KC_BTN1, KC_BTN2,
        KC_LCTL, KC_W,     KC_X,    KC_LEFT, KC_DOWN, KC_RGHT, KC_BTN3, KC_NO,
        KC_NO,   KC_NO,    KC_NO,   KC_NO,   KC_NO,   KC_NO,   KC_NO,   KC_NO
    ),
    [1] = LAYOUT(
        RGB_RMOD,RGB_MOD, RGB_HUI, RGB_VAI, KC_TRNS, RGB_SAI, RGB_HUD, RGB_VAD,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_SAD
    ),
```

_Originally posted by @noroadsleft in https://github.com/qmk/qmk_firmware/pull/11328#discussion_r574942213_

Jenpatrick, 2022-01-10:
I'd like to finish off what this person started, I've got apretty good handle, on what need to be done,t o do it.. but  if there's anything glaring that need to be done.. Let me know... I'll fix, and open a PR for this by 1/15/2022... since i have this macro KB next to me now, and am on ubuntu to debug it,.

noroadsleft, 2022-03-10:
If you have an updated branch you should open a PR.

closed 2022-07-19 by github-actions[bot]