qmk/qmk_firmware #15823: [Bug] KC_RALT sending wrong key event?

hfjn, 2022-01-10:
Hi, and thanks for this awesome piece of software.
I encountered some weird behavior with the ALTGR key and the Eurkey layout and couldn't find anyone mentioning this online.
So far I have been using a plain old keyboard and when hitting ALT GR on that one the QMK Configurator Testing function gives me:

```
KEY-DOWN - QMK: KC_RALT Event key: AltGraph    Code: AltRight      KeyCode: 225
KEY-UP   - QMK: KC_RALT Event key: AltGraph    Code: AltRight      KeyCode: 225 in 83.000ms
```

on my redox I am getting this.

```
KEY-DOWN - QMK: KC_RALT Event key: Alt         Code: AltRight      KeyCode: 18
KEY-UP   - QMK: KC_RALT Event key: Alt         Code: AltRight      KeyCode: 18 in 85.000ms
```
So it seems the sent keycode is actually not correct.
This wouldn't really bother me but sadly enough the Alt Gr key is the modifier key for the EurKey layout to access Umlauts.
Any help would be greatly appreciated. 
Cheers!

- keyboard: redox/rev1 with some minor keymap tweaks
 - Operating system: Manjaro (5.13.19-2-MANJARO)
 - AVR GCC version: 8.3.0
 - ARM GCC version: 11.2.0
 - QMK Firmware version: 0.15.2

Nuc1eoN, 2022-09-13:
Oh good catch!

That would explain why in my case the following override did not work on EurKey layout:
```c
const key_override_t home = ko_make_basic(KC_RALT, KC_LEFT, KC_HOME); // ALT_GR + Left
const key_override_t end = ko_make_basic(KC_RALT, KC_RIGHT, KC_END); // ALT_GR + Right
const key_override_t pgup = ko_make_basic(KC_RALT, KC_UP, KC_PGUP); // ALT_GR + Up
const key_override_t pgdown = ko_make_basic(KC_RALT, KC_DOWN, KC_PGDOWN); // ALT_GR + Down
```

Switching to EN layout it works.

I also wonder what the underlying issue is :thinking:

sigprof, 2022-09-13:
The original problem is most likely caused by X11 not applying the XKB settings to a hotplugged keyboard. Some desktop environments may reapply those settings automatically, but in some simpler desktop environments you may need to arrange that yourself (e.g., by invoking `setxkbmap` with all required options again); alternatively, you may move your XKB configuration to `xorg.conf` (e.g., to a file in `/etc/X11/xorg.conf.d/`) — that configuration gets applied to any hotplugged keyboards too.

------

> `const key_override_t home = ko_make_basic(KC_RALT, KC_LEFT, KC_HOME); // ALT_GR + Left`

This override is actually incorrect — the first argument needs to be a modifier mask, not a keycode. If you want to use specifically the right Alt modifier for this override, you would need to write something like this:

```c
const key_override_t home = ko_make_basic(MOD_BIT(KC_RALT), KC_LEFT, KC_HOME); // ALT_GR + Left
```

If you want to specify any Alt key, you may use `MOD_MASK_ALT`.

BTW, key overrides with the Alt modifier may not work correctly due to the host doing some undesired action when it observes what looks like a tap of the Alt key (many apps use that as the “activate the main menu” shortcut). However, using the AltGr key may actually help here, because a tap of the Alt key which is assigned the AltGr function in the OS layout won't be handled as a request to activate the menu.

Nuc1eoN, 2022-09-13:
>This override is actually incorrect — the first argument needs to be a modifier mask, not a keycode. If you want to use specifically the right Alt modifier for this override, you would need to write something like this:

Thank you yes, that was actually it. I did not know ithad tobe a modifier mask Thanks for the guidance.

> BTW, key overrides with the Alt modifier may not work correctly due to the host doing some undesired action when it observes what looks like a tap of the Alt key (many apps use that as the “activate [...]

Indeed. And that was my intention behind using Alt Gr instead of Alt. It seems to work quite well, or at least yet I havent had any collisions.