qmk/qmk_firmware #15830: [Feature Request] Rust Bindings

waynevanson, 2022-01-11:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I understand Rust much better than C or C++, so I would love to see rust bindings.
I would like to see if others have a preference for Rust over the current setup.

The end goal is for users to write their own keyboard layouts in Rust if they wish.

It's been done to a small extent before, but is currently not up to date:
- https://about.houqp.me/posts/rusty-c/
- https://github.com/houqp/qmk_firmware/tree/massdrop_houqp_rust

I'm not sure how much effort this requires but I'm going to experiment and see how it goes.

tzarc, 2022-01-11:
#7475 has the current stance on Rust.

waynevanson, 2022-01-11:
Okay, thank you.

Looks like I searched issues and not PR's.

closed 2022-01-11 by waynevanson