qmk/qmk_firmware #15843: [Bug] Since revision 0.14.20, the compilation start of `make all: all` is abnormally slow.

mtei, 2022-01-11:
## Describe the Bug

Prior to 0.14.20, compilation will start 6 seconds after submitting `make all: all`.
```
 % git checkout 0.14.20^
Previous HEAD position was 20f81af98a Allow community layouts to be specified in info.json (#14682)
HEAD is now at f72b2ad720 [Keyboard] Add Iris Rev. 6 (#14683)

 % make clean; date; make all:all
QMK Firmware 0.14.19
Deleting .build/ ... done.
Wed Jan 12 03:05:51 JST 2022
QMK Firmware 0.14.19
Making 0_sixty with keymap default
^Cmake: *** [all:all] Interrupt: 2

 % date
Wed Jan 12 03:05:57 JST 2022
```

After 0.14.20, it takes more than 7 minutes to start compiling after submitting `make all: all`.
```
 % git checkout 0.14.20          
Previous HEAD position was f72b2ad720 [Keyboard] Add Iris Rev. 6 (#14683)
HEAD is now at 20f81af98a Allow community layouts to be specified in info.json (#14682)

 % make clean; date; make all:all
QMK Firmware 0.14.20
Deleting .build/ ... done.
Wed Jan 12 03:06:32 JST 2022
QMK Firmware 0.14.20
⚠ dyz/dyz60: DEVICE_VER in config.h is overwriting usb.device_ver in info.json
⚠ dyz/dyz60: DIODE_DIRECTION in config.h is overwriting diode_direction in info.json
⚠ dyz/dyz60: RGB_DI_PIN in config.h is overwriting rgblight.pin in info.json
⚠ dyz/dyz60: RGBLED_NUM in config.h is overwriting rgblight.led_count in info.json
⚠ dyz/dyz60: PRODUCT_ID in config.h is overwriting usb.pid in info.json
⚠ dyz/dyz60: VENDOR_ID in config.h is overwriting usb.vid in info.json
⚠ dyz/dyz60: Matrix pins are specified in both info.json and config.h, the config.h values win.
☒ evyd13/atom47/rev3: evyd13/atom47/rev3: LAYOUT_full_space: Number of elements in info.json does not match! info.json:44 != LAYOUT_full_space:45
☒ evyd13/atom47/rev4: evyd13/atom47/rev4: LAYOUT_full_space: Number of elements in info.json does not match! info.json:44 != LAYOUT_full_space:45
☒ eyeohdesigns/sprh: eyeohdesigns/sprh: LAYOUT_split: Number of elements in info.json does not match! info.json:68 != LAYOUT_split:69
☒ gboards/gergoplex: MATRIX_COLS is inconsistent with the size of MATRIX_COL_PINS: 5 != 4
☒ gboards/gergoplex: MATRIX_ROWS is inconsistent with the size of MATRIX_ROW_PINS: 4 != 10
☒ lazydesigners/the60/rev2: lazydesigners/the60/rev2: LAYOUT_wkl_iso: Number of elements in info.json does not match! info.json:61 != LAYOUT_wkl_iso:64
☒ lazydesigners/the60/rev2: lazydesigners/the60/rev2: LAYOUT_hhkb_iso: Number of elements in info.json does not match! info.json:61 != LAYOUT_hhkb_iso:64
☒ melgeek/mj6xy/rev3: melgeek/mj6xy/rev3: LAYOUT_64_iso_split_bs_3u_spc: Number of elements in info.json does not match! info.json:67 != LAYOUT_64_iso_split_bs_3u_spc:66
☒ melgeek/mj6xy/rev3: melgeek/mj6xy/rev3: LAYOUT_60_iso_1u_lshift_7u_spc_split_bs_rshift: Number of elements in info.json does not match! info.json:64 != LAYOUT_60_iso_1u_lshift_7u_spc_split_bs_rshift:63
☒ Not including data from file: keyboards/onekeyco/dango40/info.json
☒ 	layouts.LAYOUT: Additional properties are not allowed ('layout_all' was unexpected)
☒ pizzakeyboards/pizza65: pizzakeyboards/pizza65: LAYOUT_65_iso_blocker_doublebs: Number of elements in info.json does not match! info.json:70 != LAYOUT_65_iso_blocker_doublebs:69
☒ projectcain/vault45: projectcain/vault45: LAYOUT: Number of elements in info.json does not match! info.json:47 != LAYOUT:46
☒ sidderskb/majbritt/rev2: sidderskb/majbritt/rev2: LAYOUT: Number of elements in info.json does not match! info.json:69 != LAYOUT:70
☒ spacey: spacey: LAYOUT: Number of elements in info.json does not match! info.json:68 != LAYOUT:70
⚠ tkw/grandiceps: MANUFACTURER in config.h is overwriting manufacturer in info.json
⚠ tkw/grandiceps/rev2: MANUFACTURER in config.h is overwriting manufacturer in info.json
☒ viktus/sp_mini: MATRIX_COLS is inconsistent with the size of MATRIX_COL_PINS: 7 != 8
Making 0_sixty with keymap default                                                                     [OK]
Making 0_sixty with keymap ven0mtr0n
^Cmake: *** [all:all] Interrupt: 2

 % date
Wed Jan 12 03:14:13 JST 2022
```

## System Information

 - QMK Firmware version: 0.14.20

tzarc, 2022-08-29:
For anyone getting to this, it's mainly as a result of the python CLI being hooked in, now.
Lots of `qmk list-keyboards` and `qmk list-keymaps`. Should probably be refactored if we want to go down this route.

Suggest a shorter "fix" would be to amend `qmk multibuild` to accept `-km all`.

tzarc, 2023-02-02:
`qmk mass-compile -j4 -km all` should work fine, now.

closed 2023-02-02 by tzarc