qmk/qmk_firmware #15849: [Feature Request] Chromatic accordion music mode

4e554c4c, 2022-01-11:
I have an OLKB Planck v4 which supports audio. There are several music modes for the Planck, however they are all quite difficult to play (in my opinion). An additional layout that provides both the benefits of a chromatic and major layout would be useful, for example the chromatic C-system accordion layout

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
The C-system layout looks like this:
[image]

Accordions, like keyboards have multiple rows. The C-system layout is designed to take advantage of this by placing (chromatically) adjacent notes diagonally instead of horizontally of each-other. Furthermore, Accordions, like keyboards, may have more or fewer rows. The minimum number of rows for the C-system is 3, and any additional rows repeat the notes of the bottom 3. This is beneficial because if you have 5+ rows you can play in any key the exact same way. I propose the C-system chromatic accordion layout be added as a music mode because this would make playing music on my keyboard a whole lot easier when I'm bored :stuck_out_tongue_closed_eyes:

4e554c4c, 2022-04-16:
not stale, still a bug, i just haven't fixed it yet

4e554c4c, 2022-07-16:
to do

tzarc, 2022-10-15:
Please open a PR when this is done -- keeping this issue open just adds noise.

closed 2022-10-15 by tzarc