qmk/qmk_firmware #15918: Keycodes组合键设置不起作用，Keycodes key combination setting doesn't work

jeromeli002, 2022-01-18:
What is the difference between Modifier and Mod-Tap? Neither MT (MOD_LCTL | MOD_LSFT | MOD_LALT, KC_K) nor MT (MOD_MEH, KC_K) will work. They can only trigger the modifier keys in front, but not the ones seen later. I need to realize how to set shift+ctrl+alt+K at the same time? 0x4D (77) can be found corresponding to M, how should the key combination find the corresponding hexadecimal

Modifier和 Mod-Tap是什么区别？MT(MOD_LCTL | MOD_LSFT | MOD_LALT, KC_K)和MT（MOD_MEH, KC_K)都不行，只能触发前面的修饰键，后见的按键不行。我需要实现同时按下 shift+ctrl+alt+K应该怎么设置？0x4D (77) 对应M可以找到，组合键应该怎么找到对应的十六进制

givedrug, 2022-02-09:
you can use the [Macros](https://docs.qmk.fm/#/feature_macros?id=macros) 
e.g.

```
enum custom_keycodes {
    SCAK = SAFE_RANGE,
};

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case SCAK:
            if (record->event.pressed) {
                // when keycode SCAK is released
                register_code(KC_LEFT_SHIFT);
                register_code(KC_LEFT_CTRL);
                register_code(KC_LEFT_ALT);
                tap_code(KC_K);
            } else {
                // when keycode SCAK is released
                unregister_code(KC_LEFT_SHIFT);
                unregister_code(KC_LEFT_CTRL);
                unregister_code(KC_LEFT_ALT);
            }
            break;
    }
    return true;
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    [0] = LAYOUT(
        SCAK,   KC_NO,  RESET,
        KC_NO,  KC_NO,  KC_NO,
        KC_NO,  KC_NO,  KC_NO
    )
};

```

givedrug, 2022-02-09:
There's another very simple way, use the MEH(KC_K)!
see more at https://docs.qmk.fm/#/feature_advanced_keycodes?id=modifier-keys

drashna, 2022-03-10:
Mod Tap is meant to by the keycode on tap, but the modifiers on hold.

Eg, `MT(MOD_MEH, KC_K)` will register *just* the mods if you hold the key longer than the tapping term, but will register *just* the K if you just tap it

closed 2022-07-13 by github-actions[bot]