qmk/qmk_firmware #15940: [Bug] Sofle right hand side encoder doesn't do anything

haarts, 2022-01-19:
I've finally managed to solder together my Keyhive version of the Sofle board. I destroyed half of the LEDs so no backlight for me. (One day someone needs to tell me how to test LEDs in the process.)

But everything works! Except for the right-hand side rotary encoder. Turning or pressing it doesn't do anything. It should emit PGDOWN and PGUP: https://github.com/qmk/qmk_firmware/blob/1f618c8302cc5053c382436a2d1fb11082c99057/keyboards/sofle/keymaps/default/keymap.c#L377-L392

(I'm using the default keymap.)

How do I debug this issue? 

(Ping @winder since he seems to know of this board?)

## System Information

 - Keyboard: Keyhive's Sofle 
   - Revision (if applicable):  rev2
 - Operating system: Mac OSX 12.1
 - AVR GCC version: avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0
 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.15.15
 - Any keyboard related software installed? No

winder, 2022-01-19:
If you have the same issue with keyhives hex file I would suspect a soldering issue. Try inspecting the joints?

haarts, 2022-01-23:
🤦 
This was the issue indeed. I'm terribly sorry to have distracted you with such a mundane issue.

Just curious: if it wasn't the soldering how would you debug this? When programming I usually sprinkle some print statements in the code, but this won't work here would it? Could you connect a terminal (like Arduino)?

closed 2022-01-23 by haarts

winder, 2022-01-23:
@haarts it was no problem, I only gave you a hint so you did the hard troubleshooting anyway!

I had a similar issues with my build, a row of keys weren't working. My problem was a broken pro micro (overheated one of the gpio pads, I think), then I overheated / broke my PCB during the replacement!

I'm not a QMK expert so can't give good advice there. Check the documentation, I think you can connect a serial port and get print statements. You could also use the OLED. Definitely a trickier than debugging something running locally!