qmk/qmk_firmware #15952: [Bug] OLED_TIMEOUT doesn't work when there's an oled animation

ach5948, 2022-01-20:
If there's an animation that regularly changes the display on an OLED screen, the `OLED_TIMEOUT` will never trigger, even if no keys are pressed.

## Describe the Bug

In the `ssd1306_sh1106` driver, `oled_task()` calls `oled_render()`, then checks the timeout and turns off the screen if the timeout is expired. `oled_render()` calls `oled_on()` if there are any updates to be made to the screen. `oled_on()` resets the timeout timer.

## System Information

 - Keyboard:  ffkeebs/puca
   - Revision (if applicable): 
 - Operating system:
 - AVR GCC version: 5.4.0

 - ARM GCC version: 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 - QMK Firmware version: develop

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

The [documentation](https://docs.qmk.fm/#/feature_oled_driver?id=basic-configuration) is a little ambiguous (just mentioning keyboard inactivity), but I would expect the oled to turn off after a certain period of time with no user activity. If this is just a misunderstanding on my part, feel free to close this ticket

filterpaper, 2022-01-21:
The documentation can probably be updated to note `after 60000ms of OLED screen update inactivity`

ricalvrod, 2022-02-07:
I think it still is a bug. For example, when using the scroll method is intended for the screen to timeout, even tho it is doing an animation. The problem is when manually doing an animation (ex: rapidly changing logos) every time the `oled_render()` is called it calls the `oled_on()` witch updates the `oled_timeout`. I think if it is not a bug when scrolling it should be here when manually animating. After all the function is for OLED Burn-in, and if the keyboard is not in use or keys being pressed it should time out.

A fix could be the removal of the `oled_on()` but it does a strange behavior when plunging in going immediately to sleep. `oled_on()` could be called inside `oled_render()` just one time at start and fixes that tho.

ach5948, 2022-02-16:
@ricalvrod While I personally agree with you, I think updating the documentation to show what the actual behavior is, is sufficient for this bug report.

closed 2022-02-16 by ach5948