qmk/qmk_firmware #16031: [Bug][develop] new addition STM32L432 introduced in #16016 has issues that bricks the board

vinorodrigues, 2022-01-25:
## Describe the Bug

thanks to @tzarc who added STM32L432 to the development branch on PR# 16016, after a failed #16012 - that in turn was a re-boot of #15050.

however this endeavour may need some more work though.

I have tried on a KC Q1 to incorporate the new code ... my steps where to:
1. pull `qmk/qmk_firmware:develop` repo.:branch 
2. and then copy across the q1 keyboards folder from the `keychron/qmk_firmware:playground` repo.:branch
3. compile with command line: `make keychron/q1/q1_ansi_stm32l432_ec11:via`

The compile process results in a `keychron_q1_q1_ansi_stm32l432_ec11_via.bin` file, though i should mention that during the compile there is a #pragma message that reads:

```
Compiling: platforms/chibios/bootloaders/none.c                                                    platforms/chibios/bootloaders/none.c:19:9: note: #pragma message: Unknown bootloader set, you may not be able to enter bootloader using software reset
 #pragma message "Unknown bootloader set, you may not be able to enter bootloader using software reset"
         ^~~~~~~
 [OK]
```
(in retro spect i should have heeded this as a stop)

4. then proceeded to flash the keeb with QMK_Toolkit - successfully flashed, and reset.  last few line were:

```
File downloaded successfully
    Transitioning to dfuMANIFEST state
*** STM32 DFU device disconnected: STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
```

this is where things go HORRIBLY wrong.  the keeb does not start up, at all.  no lights, no keypress recognition - nothing.

attempts to recover by reinstalling original f/w from vendor also do not work as the kb does not go into bootloader mode.  not with BOOTMAGIG, and also not with the reset physical switch on the PCB (there are no SPI header, so can't do that).

*

i have not looked at assumptions as to why this is in the code, generally i couldn't test them anyway as the kb is a brick now :(

## System Information

 - Keyboard:  **Keychron Q1 Knob**, a.k.a. Q1V2, a.k.a. Q1 ANSI STM32L432
   - Revision (if applicable):  **PCB V2.2 ANSI**
 - Operating system:  **macOS 10.15.7**
 - AVR GCC version: **(Homebrew AVR GCC 8.4.0_2) 8.4.0**
 - ARM GCC version: **(GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]**
 - QMK Firmware version: **0.15.16**
 - Any keyboard related software installed? **none** 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

tzarc, 2022-01-25:
Board startup works fine with my development Nucleo32-L432KC.
[image]

I'd suggest that perhaps there's been an issue with your merge or subsequent configuration?

tzarc, 2022-01-25:
Also to note, L4xx has USB DFU in ROM -- if you can get into the bootloader from a hardware pushbutton or the like, it's almost certainly recoverable.

tzarc, 2022-01-25:
Once you get into bootloader, make sure you add:

* `BOOTLOADER = stm32-dfu` to the `rules.mk`
* `VIA_ENABLE = no`

Start with the bare minimum and enable VIA once you know the base keyboard is working.
The `develop` branch does not have any support for emulated flash for L4xx, so VIA is likely to be a non-starter.

vinorodrigues, 2022-01-25:
thanks - can't tho - can't recover the PCB, it's bricked.
maybe keep this open for other to try?

tzarc, 2022-01-25:
You may _think_ it's bricked, but they're 99% recoverable if you short the correct pins when plugging it in.
Do you have a high-res photo of the PCB?

vinorodrigues, 2022-01-25:
post on here or (other)?

tzarc, 2022-01-25:
Paste here or link to imgur.com or something, doesn't matter all that much.

vinorodrigues, 2022-01-25:
Ok - try here:  https://github.com/keychron-contrib/issues/blob/main/16031/image-1.jpg

Looks like an ISP header after all - at eye-sight I thought it was an I2C header.  I can try to see if I can recover it myself but i will need to dig up my USBasp and then read up how to do that with STM32.

You shouldn't be supporting me at this level tho --- and i'm not very comfortable with you doing so --- this is a vendor issue and they should have resolved this before going to market with a "QMK" sales pitch.

By rights I should be speaking to them - but I'm fortunate to have a "my time" value larger than the cost of this thing --- so i prefer the trash can over dealing with arrogance.  Please don't bother on this unless you see something very obvious.

The only reason I reported the issue is because my a HUGE proponent of open-source and thought that my negative outcome would provide a learning for others.

tzarc, 2022-01-25:
USBasp won't help you -- that's for AVR, not for ARM.
The simplest solution will likely be this:

[image]

Ignore the text -- you want those two pins **in relation to the dot** -- your photo is already in the correct orientation, so if you short the two pins on the top side, far left, while plugging in the USB cable, it should hopefully go into bootloader mode.

tzarc, 2022-01-25:
[image]

vinorodrigues, 2022-01-25:
that did it.   but not the way i expected.

with the pins shorted the bootloader did ***not*** come in - but the qmk / keeb code ... did
but then i could do a standard reset button flash and have recovered the board!   TK!  *(that should say something tho... i'll read up the spec doc on that mcu now.)*

oh the shenanigans!  i'll try your other advice a bit later ... this resolves the bricking, but not the underlying STM32L432.

*(and did you read my note about this shouldn't be you doing this?   i had packed the thing away - needed to unpack it because i was compelled to reciprocate your time with mine :) )*

tzarc, 2022-01-25:
Good to hear!

It's not an issue -- I merged support for these MCUs so it's in my interests to make sure it's working correctly.

Not entering boot is most likely because of the missing `BOOTLOADER = stm32-dfu` as I'd previously mentioned. QMK jumps to bootloader mode if the bootmagic key is held -- without the bootloader selected no jump occurs.

I'd say try fixing that first, disabling VIA in the process.

vinorodrigues, 2022-01-25:
ok - i'll report back in a while ... hobbies don't fill the bank :)

THANKS SO MUCH for all of this ... QMK , the help, the disposition ... you're f-n awesome!

vinorodrigues, 2022-01-25:
feedback:

I can confirm that
```
BOOTLOADER = stm32-dfu
VIA_ENABLE = no
```

does indeed work 

* * *

and that
```
BOOTLOADER = stm32-dfu
VIA_ENABLE = yes
```
most defiantly *(yes, that's a play on words)* bricks it!  
and not a simple brick, its the needs 6-hands to short both PH3/BOOT0 & RST button, kinda brick!

**could it be the EEPROM emulation that #15050 attempted to add??**

ideally this KB should work with VIA.

lokher, 2022-01-26:
Actually, the BOOT pin is connected to reset button under space bar, disconnect the usb cable, hold the reset button and reconnect the usb cable would make the MCU enter boot mode. I can reproduce the issue, and I believe it is caused by lack of EEPROM emulation.

vinorodrigues, 2022-01-26:
I concur - EEPROM emulation seems to be the root cause as I've isolated everything else.  Let's see if we can get #15050 sorted to resolve this.

tzarc, 2022-01-26:
In the meantime, #16054 should prevent someone building an invalid configuration. Would be good if you can apply that PR and build with VIA enabled -- it _should_ error out, saying that there's not enough EEPROM space.

tzarc, 2022-01-26:
Following up with that PR, you can then try to add this to config.h to increase the temporary buffer to fake EEPROM:
```c
#define TRANSIENT_EEPROM_SIZE 1024
```

closed 2022-08-10 by vinorodrigues