qmk/qmk_firmware #16043: [Bug] IS_LAYER_ON doesn't work correctly in suspend_wakeup_init_user

fenuks, 2022-01-25:
## Describe the Bug
I have keymap for ghost_squid microcontroller that defines QWERTY and Colemak layouts. I switch between them with DF(), and when Colemak is a base layer, Scroll Lock diode is turned on.

In order to turn it off when computer is put to sleep, I have added
```c
void suspend_power_down_user(void) {
    writePinHigh(LED_SCROLL_LOCK_PIN);
}
```
in my keymap.c file, but that doesn't seem to be called at all. I ‘fixed’ that by moving this into [quantum.c suspend_power_down_quantum](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L490) to make it work, diode is turned off on sleep.

And finally, in order to turn on diode if needed on wakeup, I added this snippet
```c
void suspend_wakeup_init_user() {
    if (IS_LAYER_ON(KM_C)) {
        writePinLow(LED_SCROLL_LOCK_PIN);
    }

    // added for debug purposes
    if (IS_LAYER_ON(KM_Q)) {
        writePinLow(LED_CAPS_LOCK_PIN);
    }
}
```

The problem is that no matter on which base layer I am in (KM_C means Colemak, KM_Q means QUERTY which is default layer), QMK turns on Caps Lock diode, which means it thinks QWERTY layer is on, even though I can clearly tell by typing something that this is not the case. 

## System Information

 - Keyboard:  
   - Revision (if applicable):  20140518
 - Operating system: Arch Linux
 - AVR GCC version: avr-gcc (GCC) 11.2.0
 - QMK Firmware version: 0.15.17, latest git
 - Any keyboard related software installed? No

ljluestc, 2023-09-03:
```
{
    char effect[6];
    char spd[5];
    snprintf(spd, sizeof(spd), "S%d", rgblight_get_speed());
    char hue[5];
    snprintf(hue, sizeof(hue), "H%d", rgblight_get_hue());
    char sat[5];
    snprintf(sat, sizeof(sat), "C%d", rgblight_get_sat());
    char val[5];
    snprintf(val, sizeof(val), "B%d", rgblight_get_val());
    switch (rgblight_get_mode()) {
        case RGBLIGHT_MODE_STATIC_LIGHT:
            strcpy(effect, "STATC");
            break;
        case RGBLIGHT_MODE_KNIGHT:
        case RGBLIGHT_MODE_KNIGHT+1:
        case RGBLIGHT_MODE_KNIGHT+2:
            strcpy(effect, "KNIGT");
            break;
        // ... Add cases for other RGB lighting modes ...
        default:
            break;
    }
    oled_write_ln(effect, false);
    oled_write_ln(spd, false);
    oled_write_ln(hue, false);
    oled_write_ln(sat, false);
    oled_write_ln(val, false);
}
```

zvecr, 2023-09-04:
`IS_LAYER_ON` does not take into account default layers, as those are stored in a separate variable. You would need to do something like `if(layer_state_cmp(default_layer_state, KM_C))`.

closed 2023-09-04 by zvecr