qmk/qmk_firmware #16056: [Bug] HOLD-TAP for shift not working properly

jin-ahn, 2022-01-26:
HOLD-TAP keymapping does not work properly

## Describe the Bug
In my keymap i have `LSFT_T(KC_TAB)` meaning I want a key to be shift if held, tab if tapped. However when I flash it the behavior is that both tapping and holding the key gives me "shift+tab"

## System Information

 - Keyboard:  boardwalk
   - Revision (if applicable):  
 - Operating system: macos
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
keymap.c
```
#include QMK_KEYBOARD_H

enum { TD_ESC_COL = 0 };

// Tap Dance Definitions
qk_tap_dance_action_t tap_dance_actions[] = {
    // Tap once for ', twice for ESC Lock
    [TD_ESC_COL] = ACTION_TAP_DANCE_DOUBLE(KC_QUOT, KC_ESC)
    // Other declarations would go here, separated by commas, if you have them
};
const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {[0] = LAYOUT_ortho_2x2u(KC_ESC, KC_1, KC_2, KC_3, KC_4, KC_5, KC_HOME, KC_END, KC_6, KC_7, KC_8, KC_9, KC_0, KC_BSPC, KC_ESC, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_LBRC, KC_RBRC, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_BSLS, MT(MOD_LSFT, KC_TAB), KC_A, KC_S, KC_D, KC_F, KC_G, KC_EQL, KC_MINS, KC_H, KC_J, KC_K, KC_L, KC_SCLN, TD(TD_ESC_COL), KC_GRV, KC_Z, KC_X, KC_C, KC_V, KC_B, LT(3, KC_DEL), KC_TAB, KC_N, KC_M, KC_COMM, KC_DOT, LALT_T(KC_SLSH), KC_ENT, MO(5), KC_LALT, KC_LALT, KC_LGUI, KC_LCTL, LT(1, KC_BSPC), LT(2, KC_SPC), KC_RSFT, LT(5, KC_LEFT), KC_DOWN, KC_UP, KC_RGHT),
                                                              [1] = LAYOUT_ortho_2x2u(TO(0), KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F11, KC_F12, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_TRNS, KC_NO, KC_NO, LALT(KC_F), KC_NO, LALT(KC_T), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_F12, KC_DEL, KC_HOME, KC_NO, KC_PGUP, KC_PGDN, KC_END, KC_NO, KC_NO, KC_LEFT, KC_DOWN, KC_UP, KC_RGHT, KC_NO, KC_PSCR, KC_TRNS, KC_TRNS, KC_TRNS, KC_NO, LALT(KC_GRV), LALT(KC_B), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS),
                                                              [2] = LAYOUT_ortho_2x2u(TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_TRNS, KC_EXLM, KC_AT, KC_LBRC, KC_RBRC, KC_PIPE, KC_TRNS, KC_TRNS, KC_UNDS, KC_P7, KC_P8, KC_P9, KC_PAST, KC_TRNS, KC_BSPC, KC_HASH, KC_DLR, KC_LPRN, KC_RPRN, KC_AMPR, KC_TRNS, KC_TRNS, KC_MINS, KC_P4, KC_P5, KC_P6, KC_COLN, KC_TRNS, KC_TRNS, KC_PERC, KC_CIRC, KC_LCBR, KC_RCBR, KC_ASTR, KC_TRNS, KC_TRNS, KC_EQL, KC_P1, KC_P2, KC_P3, KC_PSLS, KC_TRNS, KC_NO, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_P0, KC_PPLS, KC_PDOT, KC_PCMM, KC_TRNS),
                                                              [3] = LAYOUT_ortho_2x2u(TO(0), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, RGB_VAI, RGB_SAI, RGB_SPI, KC_NO, KC_NO, KC_NO, RESET, KC_NO, KC_NO, KC_NO, KC_MS_U, KC_NO, KC_WH_U, KC_NO, RGB_VAD, RGB_SAD, RGB_SPD, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_MS_L, KC_MS_D, KC_MS_R, KC_WH_D, KC_NO, RGB_MOD, RGB_HUI, KC_NO, KC_NO, KC_NO, KC_NO, KC_MPLY, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_BTN2, KC_NO, RGB_RMOD, RGB_HUD, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(6), KC_NO, KC_NO, KC_BTN5, KC_BTN4, KC_BTN1, RGB_TOG, KC_MUTE, KC_MRWD, KC_VOLD, KC_VOLU, KC_MFFD),
                                                              [4] = LAYOUT_ortho_2x2u(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_SLEP, KC_NO, MEH(KC_INS), MEH(KC_UP), MEH(KC_HOME), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MEH(KC_LEFT), MEH(KC_DOWN), MEH(KC_RGHT), LCAG(KC_Z), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MEH(KC_DEL), LCAG(KC_C), MEH(KC_PGDN), LCAG(KC_X), LCAG(KC_V), LALT(KC_F4), LCA(KC_DEL), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, LCAG(KC_LEFT), LCAG(KC_RGHT), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO),
                                                              [5] = LAYOUT_ortho_2x2u(LALT(KC_F4), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_SLEP, KC_TRNS, LCAG(KC_MINS), LGUI(KC_UP), LCAG(KC_EQL), LCAG(KC_6), LCAG(KC_9), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LCA_T(KC_DEL), LGUI(KC_LEFT), LGUI(KC_DOWN), LGUI(KC_RGHT), LCAG(KC_5), LCAG(KC_8), KC_TRNS, KC_TRNS, KC_LEFT, KC_TRNS, OSM(MOD_LCTL | MOD_LGUI), KC_RGHT, KC_TRNS, KC_NO, KC_TRNS, LCAG(KC_1), LCAG(KC_2), LCAG(KC_3), LCAG(KC_4), LCAG(KC_7), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, LCAG(KC_COMM), LCAG(KC_DOT), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, SGUI(KC_RGHT), LCA(KC_SLSH), LCA(KC_SCLN), KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS),
                                                              [6] = LAYOUT_ortho_2x2u(KC_ESC, KC_K, KC_B, KC_O, KC_L, KC_F5, KC_SCLN, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TO(0), KC_DOT, KC_TAB, KC_2, KC_W, KC_3, KC_4, LCTL(KC_T), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_M, KC_LSFT, KC_A, KC_S, KC_D, KC_5, KC_T, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_MINS, KC_6, KC_F5, KC_R, KC_7, KC_8, KC_0, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_H, KC_Y, KC_G, KC_9, KC_SPC, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)};
```

config.h
```
#pragma once

#include "config_common.h"

/* USB Device descriptor parameter */
#define VENDOR_ID 0xCDCD
#define PRODUCT_ID 0x5337
#define DEVICE_VER 0x0001
#define MANUFACTURER shensmobile
#define PRODUCT Boardwalk

/* key matrix size */
#define MATRIX_ROWS 5
#define MATRIX_COLS 14

/*
 * Keyboard Matrix Assignments
 *
 * Change this to how you wired your keyboard
 * COLS: AVR pins used for columns, left to right
 * ROWS: AVR pins used for rows, top to bottom
 * DIODE_DIRECTION: COL2ROW = COL = Anode (+), ROW = Cathode (-, marked on diode)
 *                  ROW2COL = ROW = Anode (+), COL = Cathode (-, marked on diode)
 *
 */
#define MATRIX_ROW_PINS \
    { F0, F1, F4, F5, F6 }
#define MATRIX_COL_PINS \
    { F7, C7, C6, B6, B5, B4, D7, D6, D4, D5, D3, D2, D1, D0 }
#define UNUSED_PINS

/* COL2ROW, ROW2COL */
#define DIODE_DIRECTION COL2ROW

// #define BACKLIGHT_PIN F5
// #define BACKLIGHT_LEVELS 6

/* Debounce reduces chatter (unintended double-presses) - set 0 if debouncing is not needed */
#define DEBOUNCE 5

/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE
/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE

/*
 * Feature disable options
 *  These options are also useful to firmware size reduction.
 */

/* disable debug print */
//#define NO_DEBUG

/* disable print */
//#define NO_PRINT

/* disable action features */
//#define NO_ACTION_LAYER
//#define NO_ACTION_TAPPING
//#define NO_ACTION_ONESHOT
//#define NO_ACTION_MACRO
//#define NO_ACTION_FUNCTION

#define TAPPING_TERM 200

// ws2812 options
#define RGB_DI_PIN B7         // pin the DI on the ws2812 is hooked-up to
#define RGBLIGHT_ANIMATIONS   // run RGB animations
#define RGBLED_NUM 14         // number of LEDs
#define RGBLIGHT_HUE_STEP 12  // units to step when in/decreasing hue
#define RGBLIGHT_SAT_STEP 25  // units to step when in/decresing saturation
#define RGBLIGHT_VAL_STEP 12  // units to step when in/decreasing value (brightness)

```

dlucenad, 2022-02-28:
I have a similar issue, in my case I have LSFT_T(KC_UNDS), but when I tap the shift key I have "-" instead of "_"

dlucenad, 2022-02-28:
I just noticed that issue [#642]((https://github.com/qmk/qmk_firmware/issues/642)) is similar to this problem.

dlucenad, 2022-02-28:
So, I was reading [the docs](https://docs.qmk.fm/#/feature_space_cadet?id=configuration) and found out that you can redefine the default keys for Space Cadet and it solved my issue, this might help you too. I think Space Cadet was interfering in the usage of LSFT_T(kc) for some reason. The way I did was I added in my config.h the following line:

`#define LSPO_KEYS KC_LSFT, KC_LSFT, KC_MINS`

And for this to work in my keymap.c I have LSPO mapped as my Left Shift. This means that when LSPO is held down it sends KC_LSFT and when taped it sends KC_MINS with modifier KC_LSFT resulting in KC_UNDS. I can't use KC_UNDS directly as it is not a [basic keycode](https://docs.qmk.fm/#/keycodes_basic), thus not supported - although the documentation only talks about [this limitation in Mod-Tap](https://docs.qmk.fm/#/mod_tap?id=caveats). 

For your case I tested here and you have to add the following line to your config.h:

`#define LSPO_KEYS KC_LSFT, KC_TRNS, KC_TAB`

closed 2023-08-03 by jin-ahn